hbldh/bleak #128: Scan Response not provided by discover()

valerianrusu, 2019-11-13:
* bleak version: 0.51
* Python version: 3.7.4
* Operating System: Windows 10

I have a BLE device that provides extra information (extra ManufacturerData) in the Scan Response message. I did not see how to get this data using discover() therefore I have modified this function to return Manufacturer data from Scan Response message as part of metadata. Is it possible that future releases will provide this feature? Maybe even provide the  the entire Scan Response event as BluetoothLEAdvertisement same as Advertise event provided currently under details. Maybe details should contain both events.

Here is the section of code containing the change that I did:

        data = {}
        for m in d.Advertisement.ManufacturerData:
            md = IBuffer(m.Data)
            b = Array.CreateInstance(Byte, md.Length)
            reader = DataReader.FromBuffer(md)
            reader.ReadBytes(b)
            data[m.CompanyId] = bytes(b)

        dataScan = {}
        for sr in scan_responses.values():
            if d.BluetoothAddress == sr.BluetoothAddress:
                for m in sr.Advertisement.ManufacturerData:
                    md = IBuffer(m.Data)
                    b = Array.CreateInstance(Byte, md.Length)
                    reader = DataReader.FromBuffer(md)
                    reader.ReadBytes(b)
                    dataScan[m.CompanyId] = bytes(b)
                break

        local_name = d.Advertisement.LocalName
        if not local_name and d.BluetoothAddress in scan_responses:
            local_name = scan_responses[d.BluetoothAddress].Advertisement.LocalName
        found.append(
            BLEDevice(
                bdaddr,
                local_name,
                d,
                uuids=uuids,
                manufacturer_data=data,
                scan_manuf_data=dataScan,
            )
        )

hbldh, 2019-12-27:
Interesting. This is a desireable improvement. I will try to analyse this during the first days of the next year.

nataddrho, 2020-03-25:
I also am developing hardware with manufacturer data in the scan response, and I need this functionality. Do you know when it will be ready?

hbldh, 2020-03-27:
It will not be done in the foreseeable future I am afraid.  I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

hbldh, 2020-09-25:
This can be done like it is done in #118.

closed 2020-09-25 by hbldh