hbldh/bleak #129: wrong version requirement for BlueZ on Linux

afflux, 2019-11-17:
Since 94163fa1d4929b243311988c420b2bcffb204270 (#61), `write_gatt_char` requires at least BlueZ 5.46, because `AcquireWrite` was not introduced before.

The current version checking logic in [`__init__.py`](https://github.com/hbldh/bleak/blob/11e7adf1ae1774b22667fcc4e6533ad69597951a/bleak/__init__.py#L41) and the readme files all claim that 5.43 is sufficient.

dlech, 2019-12-20:
To clarify, BlueZ 5.46 is only required to be able to "Write without response".

hbldh, 2019-12-27:
Documentation in v0.5.2 of Bleak will provide clearer instructions regarding BlueZ version requirements. Thank you for requesting clearer specification of this.

As @dlech says,  the version requirements is higher than 5.43 only in the case of the `write_gatt_char` method, and only if the "Write without response" is desired.

hbldh, 2020-01-05:
Adressed and fixed in #138, and further in commit b767f9d

afflux, 2020-01-06:
I'd argue that the default `response=False` did not really make sense, because that's the stricter version requirement - but then again, with the exception in #138, chances are higher for people to find the parameter.

closed 2020-01-06 by afflux

hbldh, 2020-01-06:
@afflux In my mind, the `response=False` was the simpler choice and thus the default behaviour. The other OS backends have no such version requirement demands, and I don't want to change default behaviour for all of them. 

It is now a regrettable byproduct of a choice I made in the advent of this project.