hbldh/bleak #130: BlueZ bus interface is not cached and can not be cleaned up properly

afflux, 2019-11-17:
* bleak version: 0.5.1
* Python version: 3.5
* Operating System: osmc
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.43

### Description

Repeated calls to `client.connect()` (both on the same instance, as well as newly created instances) will create new bus-objects (see [client.py:96](https://github.com/hbldh/bleak/blob/11e7adf1ae1774b22667fcc4e6533ad69597951a/bleak/backends/bluezdbus/client.py#L96)) for each call, without the means to clean them up again.

### What I Did
Auto-Reconnect loop like this:

```python
while True:
    client = bleak.BleakClient(addr, loop=loop)
    try:
        await client.connect()
    except bleak.exc.BleakError as exc:
        logging.error('failed to connect: %s', exc)
        continue
    [... logic here ....]
```

### What I expected
Either `client.__del__` or an explicit `client.cleanup` to call `client._bus.disconnect`, or `client._bus` to be cached.

### What happened
```
WARNING failed to connect: org.bluez.Error.Failed: Software caused connection abort
[previous message repeats 244 times]
Traceback (most recent call last):
  File "/usr/local/sbin/ble.py", line 156, in reconnect
    await client.connect()
  File "/opt/bleak-venv/lib/python3.5/site-packages/bleak/backends/bluezdbus/client.py", line 93, in connect
    await discover(timeout=timeout, device=self.device, loop=self.loop)
  File "/opt/bleak-venv/lib/python3.5/site-packages/bleak/backends/bluezdbus/discovery.py", line 128, in discover
    bus = await client.connect(reactor, "system").asFuture(loop)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
txdbus.error.RemoteError: org.freedesktop.DBus.Error.LimitsExceeded: The maximum number of active connections for UID 0 has been reached
```

hbldh, 2019-12-27:
I would recommend you to use the `async with` context manager solution instead of creating an object and never disconnecting it:

```python
while True:
    async with BleakClient(mac_addr, loop=loop) as client:
        [... logic here ....]
```

At least try to do this instead:

```python
while True:
    client = bleak.BleakClient(addr, loop=loop)
    try:
        await client.connect()
    except bleak.exc.BleakError as exc:
        logging.error('failed to connect: %s', exc)
        continue
    finally:
        await client.disconnect()
    [... logic here ....]
```

The disconnect method has a cleanup which it runs when disconnect is called.

Try this and please report back.

afflux, 2019-12-28:
It's not about the DBUS org.bluez.Device1 Disconnect method, but the txdbus.Bus instance. Looking at the core, the client.disconnect() function wouldn't help me, as the txdbus.Bus is still alive and connected after client.disconnect(). (FWIW, the exception listed above happens even when the DBUS org.bluez.Device1 Connect call fails for every client instance created)

An option to close the txdbus Bus on the bleak.backends.bluezdbus.client.Client object, or to be able to re-use the same _bus over multiple bleak.backends.bluezdbus.client.Client instances would solve the issue

hbldh, 2020-01-06:
Ok. I think I see the problem now. I use a global Twisted reactor for creating the DBus System Bus connection, and I never call `stop` on that reactor since other `BleakClient`s are potentially depenmdent on it. I will explore whether or not a local reactor can be used or if the bus connection should be moved outside of the `BleakClient`.

hbldh, 2020-01-06:
This will be most åprobably be solved by #143.

closed 2020-03-09 by hbldh

afflux, 2020-04-21:
> This will be most probably be solved by #143.

I had the chance to test this only now. Unfortunately, this issue was only shifted to a different place.
The AsyncioSelectorReactor creates a new pipe for each instance, which is never cleaned up. Simple script to see the issue:

```
from twisted.internet.asyncioreactor import AsyncioSelectorReactor

while True:
    reactor = AsyncioSelectorReactor()
```

```
Traceback (most recent call last):
  File "test.py", line 5, in <module>
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/asyncioreactor.py", line 69, in __init__
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/base.py", line 571, in __init__
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 284, in installWaker
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 133, in __init__
OSError: [Errno 24] Too many open files
```

And no, spinning up and stopping the reactor doesn't help:

```
from twisted.internet.asyncioreactor import AsyncioSelectorReactor

while True:
    reactor = AsyncioSelectorReactor()
    reactor.callLater(.001, reactor.stop)
    reactor.run()
    print('reactor stopped')
```

```
Unhandled Error
Traceback (most recent call last):
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/base.py", line 434, in fireEvent
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/defer.py", line 322, in addCallback
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/defer.py", line 311, in addCallbacks
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/defer.py", line 654, in _runCallbacks
    
--- <exception caught here> ---
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/base.py", line 447, in _continueFiring
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/base.py", line 1278, in _reallyStartRunning
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 298, in _handleSignals
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 200, in __init__
    
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 133, in __init__
    
builtins.OSError: [Errno 24] Too many open files

reactor stopped
Traceback (most recent call last):
  File "test.py", line 5, in <module>
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/asyncioreactor.py", line 69, in __init__
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/base.py", line 571, in __init__
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 284, in installWaker
  File "/tmp/twisted-env/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 133, in __init__
OSError: [Errno 24] Too many open files
```

My system has a `ulimit -d` setting of 1024, so as soon as 102x-something connect() calls were performed, my application will crash.

(I do understand that I need to raise this issue with the twisted team - I'm just documenting this here for the record.)

afflux, 2020-04-21:
Just saw the [comments on the PR](https://github.com/hbldh/bleak/pull/143#issuecomment-582103954) as well as issue #169. I think this is adressed by PR #154. Thanks.

commit afcfff7b (2020-01-06): Residual changes for #143.

This tries to ensure that reactor is closed and deleted after client disconnects, to fix #130
commit b2710d68 (2020-01-06): Residual changes for #143.

This tries to ensure that reactor is closed and deleted after client disconnects, to fix #130
commit d1a73bdb (2020-04-06): bluez: Cleanup all resources even if the client disconnects

Recently there have been changes to make sure we clean up all DBus and
Twisted resources when we disconnect from a server to prevent memory
leaks and running out of available DBus connections. However, this was
only implemented for the case where the application actively disconnects
form the server. If the disconnection was initiated by the server for
example or by going out of range, the dbus connections still leaked.

This patch splits the cleanup into two different methods. One for
cleaning the subscriptions to DBus messages (for  [... cut]
PR #193 (2020-05-15): Version 0.6.2
* Better cleanup of Bluez notifications (merges #154 and fixes #130 )
* Fix for ``read_gatt_char`` in Core Bluetooth (fixes #177)
* Fix for ``is_disconnected`` in Core Bluetooth (merges #187 and fixes #185)
* Added ``disconnection_callback`` functionality for Core Bluetooth (merges #186 and fixes #184)
* Documentation fixes
* Added ``requirements.txt``