hbldh/bleak #131: Not reacting to Advertiser with SCAN_REQ

BlackHawk1912, 2019-11-21:
* bleak version: 0.5.1
* Python version: 3.6.4
* Operating System: Windows 10 Build 17763

### Description
I ran the service-explorer.py example script and sometimes it works but in about 80% of my connection attempts I get the error 
```
Traceback (most recent call last):
  File "c:/Users/VOS9LOL/WSL/Libraries/bleak-master/examples/service_explorer.py", line 64, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Program Files\Anaconda3\lib\asyncio\base_events.py", line 467, in run_until_complete
    return future.result()
  File "c:/Users/VOS9LOL/WSL/Libraries/bleak-master/examples/service_explorer.py", line 29, in run
    async with BleakClient(address, loop=loop) as client:
  File "C:\Program Files\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Program Files\Anaconda3\lib\site-packages\bleak\backends\client.py", line 43, in __aenter__
    await self.connect()
  File "C:\Program Files\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Program Files\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 118, in connect
    "Device with address {0} was " "not found.".format(self.address)
bleak.exc.BleakError: Device with address D9:8B:2A:FB:62:B0 was not found.

```
Even though my devices is already paired to my windows machine through the windows settings. So I started analyzing the problem. With a Bluetooth sniffer I found that the device I want to connect to is advertising on all three channels every two seconds wiht **ADV_IND**.
When I start the script, my sniffer tells my that my windows machine is responding with **SCAN_REQ** to a lot of different devices sending ADV_IND but NOT the device I want to connect to. Even though the timeout of Bleak is set to **10 seconds** which should always find the device that advertises every **two seconds**. But it doesn't.

Different story when I set my device in pairing mode. Then it will advertise ever 10ms and the connection will work 100% of the time.

So it seems like Bleak is not able to respond to a slow advertiser, right?

hbldh, 2020-01-05:
I have heard from several users that long advertisement intervals have caused problems, but I have no solution for it.

That kind of response behaviour (ADV_IND as such) is not handled by bleak at all, but is left to the OS Bluetooth stack to handle. Bleutooth sniffers might work very different than Bleak, depending on whether it uses the OS native methods or not; Bleak cannot prodive identical or even similar behaviour in such cases.

I am planning on adding a new type of discover method to Bleak, maybe that will solve your problem.

hbldh, 2020-01-05:
Similar to #72

hbldh, 2020-09-25:
This is probably solved in version 0.8.0. Will close for now.

closed 2020-09-25 by hbldh