hbldh/bleak #132: Subscribing to a specific service

QuestionMarque, 2019-11-22:
- bleak version 0.5.1
- Python version 3.7
- Windows version 10

I have a Bluetooth client with the following four services:

`00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile`    
`00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile `    
`0000e7c0-1212-efde-1523-785fef13d123: Custome Profile `
`00001530-1212-efde-1523-785feabcd123: Unknown`

The `Custome Profile` has three characteristics (COM, Stream, Info) and I try to subscribe to COM using `com_characteristic_uuid`. However, I cannot figure out how to call a specific service. In fact by:

```python
await client.start_notify(com_characteristic_uuid, notification_handler)
```

one will pass through all characteristics of all existing services, and if there are two similar `uuid `for different services, chances are that we do not get the expected behavior. Is there any API available to subscribe to a specific service?

hbldh, 2020-01-05:
You get the correct uuid from the Service object and send that into the `start_notify` metod. See [service_explorer.py](https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py) example.

closed 2020-01-05 by hbldh

ghazijallouli, 2022-03-11:
is the start_notify the same as subscribe?

hbldh, 2022-03-12:
@ghazijallouli Yes. 

If you have general questions about Bleak, please pose these in the Discussions section instead of in old, closed issues.