hbldh/bleak #133: disconnect() doesn't disconnect from device

chunghaw-tseng, 2019-12-05:
* bleak version: 0.5.1
* Python version: 3.6
* Operating System: Windows 10

### Description

I am currently writing an application to connect to a BLE device, write to some characteristics and then disconnect from the device. From our tests, bleak's disconnect API doesn't always send the disconnect command to the device (seen using a snifer). There are cases that the devices remain connected after calling disconnect but bleak returns that they are already disconnected. Not sure where the problem might be, is there any patch for this problem?

### What I Did

I tried to disconnect, the return is True but looking at the BLE snifer the devices still transmit empty PDUs to maintain the connection. Calling self.client.is_connected() returns False. The connection has not been terminated.

```python
        try:
             response = await self.client.disconnect()  #self.client is the BleakClient 
             print("DISCONNECT RESPONSE", response)
        except BleakError:
              log.error("Device couldn't be disconnected")
        return response  #Returns true
```

hbldh, 2020-01-05:
Bleak uses, via pythonnet, the C# UWP libraries for connecting to Bluetooth devices. 
I have followed the [Universal Windows Platform (UWP) app samples](https://github.com/microsoft/Windows-universal-samples/blob/master/Samples/BluetoothLE/cs/Scenario2_Client.xaml.cs#L78-L97) method of disconnecting from a BluetoothLEDevice. That being said, it might be that Bleak, pythonnet and/or the C# bridge I have developed delay the disconnecting.

The Dispose method will eventually close the connection. There is a [`Close` method](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice.close#Windows_Devices_Bluetooth_BluetoothLEDevice_Close) that can be called, but I decided against it to avoid potential errors.

What is the real problem of the device not being freed immediately?

mitchmccarthy, 2020-01-09:
The peripheral device is intended to drop to a low-power mode after the disconnection, but this issue is causing the connection to persist even though the bleak client says it has disconnected.

We set up a BLE sniffer while we reproduced the issue, and there doesn't seem to be any TERMINATE signals sent over the air.

hbldh, 2020-01-09:
@mitchmccarthy This SO issue descibes to problems Bleak faces as well: https://stackoverflow.com/questions/39599252/windows-ble-uwp-disconnect

I would like to be able to close it, but something seems to be left in the `BluetoothLEDevice` object which makes garbage collection not collecting it. AS to the timeframe, I have no idea. Not more than a minute?

If the process in which Bleak was run is killed, is the connection severed then? Or is it still present?

If you insert specific handling in your device due to Bleak's problems, then maybe this project isn't for you. There are other BLE projects which are production ready; I do not guarantee any functionality in Bleak; it is a spare time exploration project of mine.

mitchmccarthy, 2020-01-09:
If the process (python shell) is killed then yeah the disconnect occurs pretty much immediately, so I think you're right about some hanging reference. 

We've already looked at other options, your project was very simple to drop into our application and aside from this issue it worked great so we were hoping to continue with it. I totally understand that this is a work-in-progress project, and thanks for taking the time to investigate the issue.

hbldh, 2020-09-25:
This is probably solved in version 0.8.0. Will close for now.

closed 2020-09-25 by hbldh

dlech, 2021-04-08:
Related info:

https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/9eae39ff-f6ca-4aa9-adaf-97450f2b4a6c/disconnect-bluetooth-low-energy

I'm currently having trouble getting a device to disconnect on Windows 10 with Bleak 0.11.0.

dlech, 2021-04-08:
And rebooting the computer seems to have fixed the problem.

megholm, 2022-11-30:
> If the process (python shell) is killed then yeah the disconnect occurs pretty much immediately, so I think you're right about some hanging reference.

I'll second that.
(Windows 10, Bleak 0.19.5)