hbldh/bleak #135: AttributeError: 'NoneType' object has no attribute 'obj' when subscribing to service

baldengineer, 2019-12-07:
* bleak version: 0.5.1
* Python version: 3.7.4
* Operating System: Windows  10.0.17763.864

### Description
Was trying to connect to the generic attribute profile of my device. It sends out a string, I would like to capture that. I was hoping the `enable_notifications.py` example would capture that string. However, it fails with the error:  `AttributeError: 'NoneType' object has no attribute 'obj'.`

### What I Did
Using the `enable_notifications.py` example, I changed the `CHARACTERISTIC_UUID` and `address` variables, nothing else. My device has 4 services, I get the same behavior on all four. For the record, from `getServices` example:
```
:00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
0000180a-0000-1000-8000-00805f9b34fb: Device Information
0000fff0-0000-1000-8000-00805f9b34fb: Vendor specific
00010203-0405-0607-0809-0a0b0c0d1911: Unknown
```

When I use `gatttool` on another machine (that is bluez) compatible, I connect to the MAC address and I get a stream of messages like this:

```
[A5:B3:C2:22:14:D2][LE]> connect
Attempting to connect to A5:B3:C2:22:14:D2
Connection successful
Notification handle = 0x001b value: 37 f1 04 00 ff 7f
Notification handle = 0x001b value: 37 f1 04 00 ff 7f
Notification handle = 0x001b value: 37 f1 04 00 ff 7f
[A5:B3:C2:22:14:D2][LE]> disconnect
```
I'm after that the string `37 f1 04 00 ff 7f`. It contains status information, which I have static for now so it is easier to find.

Looking around with Bluetooth LE Lab, it looks like that value is in the characteristic that matches "Generic Attribute Profile."  So I was expecting (or hoping) that the example would provide a similar string.

Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
Removed unrelated MACs and some identifying information in the paths
```
Received A5:B3:C2:22:14:D2: BDM.
Received A5:B3:C2:22:14:D2: BDM.
# Removed unrelated MACs...
Received A5:B3:C2:22:14:D2: BDM.
Received A5:B3:C2:22:14:D2: BDM.
# Removed unrelated MACs...
poll 2000.000 ms took 2000.000 ms: timeout
Could not remove event handlers: unknown event handler...
15 devices found. Watcher status: 3.
Connecting to BLE device @ A5:B3:C2:22:14:D2
poll took 78.000 ms: 1 events
Get Services...
poll took 63.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 15.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 32.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 15.000 ms: 1 events
poll took 32.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 31.000 ms: 1 events
poll took 0.000 ms: 1 events
Connection successful.
Connected: False
Disconnecting from BLE device...
Traceback (most recent call last):
  File "bleak_service_subscribe.py", line 46, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "bleak_service_subscribe.py", line 31, in run
    await client.start_notify(CHARACTERISTIC_UUID, notification_handler)
  File "\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 459, in start_notify
    status = await self._start_notify(characteristic.obj, callback)
AttributeError: 'NoneType' object has no attribute 'obj'
[Finished in 3.1s with exit code 1]
```

baldengineer, 2019-12-20:
I was using the wrong UUID. Eventually found the notify UUID was `0000fff4-0000-1000-8000-00805f9b34fb`. The `000ff4` prefix might be unique to this device, but the suffix ending with `...34fb` seems common? Anyway, the no attribute 'obj' seems to get thrown if the UUID does not return anything useful.

closed 2019-12-20 by baldengineer

hbldh, 2019-12-27:
Thank you for logging this. I will improve the error handling regarding this.