hbldh/bleak #136: Stay connected not working

sonus89, 2019-12-08:
* bleak version: 0.5.0
* Python version:  3.7.4
* Operating System: Windows 10 Pro ( 10.0.18362 N/A Build 18362 )
* BlueZ version (`bluetoothctl -v`) in case of Linux: 0.22

### Description

I am trying to write a python script which can receive data from a BLE device.
Unfortunately I can only receive data in every 5 sec because it keeps reconnecting.
How can I write a python script which stays connected and makes me able to get data quickly? 
( As a stream )

### What I Did

    async def TxRx_BLE_callback(self):
        string = self.message + '\r\n'
        arr = bytearray(string, 'utf-8')
        RX_CHAR_UUID = "6e400002-b5a3-f393-e0a9-e50e24dcca9e"
        TX_CHAR_UUID = "6e400003-b5a3-f393-e0a9-e50e24dcca9e"
        async with BleakClient(self.selected_mac, loop=asyncio.get_event_loop()) as client:
            await client.start_notify(TX_CHAR_UUID, arr)
            await client.write_gatt_char(RX_CHAR_UUID, arr)
            self.response = await client.read_gatt_char(TX_CHAR_UUID)
            print('\tRequest:\t',string, end='')
            print('\tResponse:\t',self.response.decode('utf-8'))

    def TxRx_BLE(self):
        t4 = Thread(target=asyncio.run, args=(self.TxRx_BLE_callback(),))
        t4.start()
        t4.join()
        self.twindow.insert(tk.END, '\nDevice: ' + self.response.decode('utf-8'))
        return t4

I call the synch function ( TxRx_BLE ) from my code and it is very slow. ( 5 sec waiting every time I call it because it reconnects, reads data, disconnects .. and I don't want it to disconnect )

hbldh, 2020-01-05:
First of all, I recommend not running it in a Thread. Use the event loop in the asyncio fashion. here is a good primer: https://realpython.com/async-io-python/
Two, make sure you keep the `client` alive, since as soon as yo probram exits the `async with` scope it will get disconnected.
Three, the `start_notify` method should not be sent the `arr` object, but a method that will be used as callback-

Look at the example files and try to use that.

closed 2020-06-02 by hbldh

bua02, 2024-04-18:
> First of all, I recommend not running it in a Thread. Use the event loop in the asyncio fashion. here is a good primer: https://realpython.com/async-io-python/ Two, make sure you keep the `client` [...]

hi,

I am new to python and i am not quite sure how to "keep the client alive". What would be a good way to do that? I only need it to listen to Notifications and log them. The subscribing works quite well but it disconnects afterwards.

dlech, 2024-04-18:
See the wait for disconnect example.