hbldh/bleak #137: write_gatt_char error. correct bluez version check?

5frank, 2019-12-20:
* bleak: 0.5.1
* python: 3.5.3
* bluez: 5.43
* os: Linux-4.9.0-11-amd64-x86_64-with-debian-9.11

### Description

`write_gatt_char` (repsonse=False) fails with the following error `txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownMethod: Method "AcquireWrite" with signature "a{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist`.
The following change resolvs the issue:

```diff
diff --git a/bleak/backends/bluezdbus/client.py b/bleak/backends/bluezdbus/client.py
index ac779f8..3585659 100644
--- a/bleak/backends/bluezdbus/client.py
+++ b/bleak/backends/bluezdbus/client.py
@@ -399,7 +399,7 @@ class BleakClientBlueZDBus(BaseBleakClient):
                 % str(_uuid)
             )
 
-        if response or (self._bluez_version[0] == 5 and self._bluez_version[1] > 50):
+        if response or (self._bluez_version[0] == 5 and self._bluez_version[1] > 42):
             # TODO: Add OnValueUpdated handler for response=True?
             await self._bus.callRemote(
                 characteristic.path,
```

Is the check on bluez version correct? or perhaps debian have some patches in the bluez 5.43 version that causes this to work?

dlech, 2019-12-20:
The check in your diff is correct because it is for support for the `type` option of the `WriteValue` method, not the `AcquireWrite` that is causing your error.

`AcquireWrite` was added in Bluez 5.46. So in older Bluez versions, it is simply not possible to write without response.

5frank, 2019-12-20:
ok thanks! It all makes sense now! :)

closed 2019-12-20 by 5frank