hbldh/bleak #140: BLE MAC address on MacOS instead of system ID

5frank, 2020-01-02:
* bleak: 0.5.1
* os: Darwin-18.7.0-x86_64-i386-64bit
* python: 3.7.5

Perhaps it is only me, but it would be nice if the CoreBluetooth backend used BLE MAC address and not the system "device UUID"? It would make my scripts and configurations portable (I am usually on Linux).
I made an attempt to implement this [here](https://github.com/5frank/bleak/commit/e21384a857548314de8146f3aa2ee09699e6aa71) inspired from [noble](https://github.com/noble/noble). Initial tests seems to work even if the solution feels somewhat hacky.

Question:
1. Do anyone see any problems with this method of obtaining the address (i.e. read the "CoreBluetoothCache" from `/Library/Preferences/com.apple.Bluetooth.plist`)? Is there a better way?

2. It seems the reason MacOS uses UUID is because some security feature in BLE that hides the MAC address. Does anyone know how this is handled on Windows and Linux/BlueZ?

3. Is it a bad or good idea to accept both MacOS "device UUID" and BLE MAC address? Would make this change backward compatible, but not sure if anyone cares.

5frank, 2020-01-04:
1. It seems that CoreBluetoothCache is only updated after a connect. So you still need the UUID or some other workaround.

3. Could possible check whatever the "address" passed to `connect` is a 128-bit UUID or 48-bit MAC address an handle it accordingly. (assuming there is a sane way to obtain the MAC)

hbldh, 2020-01-05:
From the Bleak documentation:
> The most noticible difference between the other backends of bleak and this backend, is that CoreBluetooth doesn’t scan for other devices via MAC address. Instead, UUIDs are utilized that are often unique between the device that is scanning the the device that is being scanned.

It would be nice to be able to use MAC address, but the effort to circumvent the macOS UUID solution is not proportional to the benefits. Reading from a cache file on disk is a reasonable idea, but I do not think it is reliable enough to be used for creating connections.

Bleak strives to have an identical API, but is accepting of OS specific differences. macOS discovers by UUID and connects by UUID, it would be counterproductive to force MAC connecting. This is one of the few "this we just have to accept" differences, in my eyes.

hbldh, 2020-06-02:
I do not think this is solvable. Will close for now.

closed 2020-06-02 by hbldh

dlech, 2022-10-07:
FYI, I have found a hack for this. See #1073. Give it a 👍 if this is still a needed feature.