hbldh/bleak #142: CoreBluetooth AdvertisementData should be handled in callback!?

5frank, 2020-01-04:
* bleak: 0.5.1
* python: 3.7.5
* os: macOS 10.14.6 (Darwin-18.7.0-x86_64-i386-64bit)

### Description
printing AdvertisementData keys in the `centralManager_didDiscoverPeripheral_advertisementData_RSSI_` callback: 
```
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataIsConnectable
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataLocalName
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataChannel
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataIsConnectable
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataServiceUUIDs
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:184: advDataKey: kCBAdvDataChannel
```
but later in discovery, printing the same keys for same device shows some of them disappeared:
```
DEBUG:bleak.backends.corebluetooth.discovery:126: advDataKey: kCBAdvDataIsConnectable
DEBUG:bleak.backends.corebluetooth.discovery:126: advDataKey: kCBAdvDataLocalName
DEBUG:bleak.backends.corebluetooth.discovery:126: advDataKey: kCBAdvDataChannel
```
i.e. AdvertisementData, RSSI etc should be handled in the callback and copied.

hbldh, 2020-01-05:
I do not own a mac myself, so the fixes for macOS issues happen not so very often. It would be best for you to implement this yourself and send a PR if it is important. 

Would it suffice to convert the `advertisementData: NSDictionary` input to a Python dict and store that in `self.advertisement_data_list` here: [`CentralManagerDelegate.py#L173`](https://github.com/hbldh/bleak/blob/master/bleak/backends/corebluetooth/CentralManagerDelegate.py#L173)?

5frank, 2020-01-05:
working on a PR. Found some other issues while debugging this. For example that devices are never removed only added.

5frank, 2020-01-05:
Half related changes to the `BLEDevice` class that I would appreciate some help with (if accepted):

* Move implementation of the class to the backends. Seems to contain backend specific code already and I do not dare to touch it.

* Add more getters for common data like service uuids, manufacturer, more?. Currently only exists for rssi. Would make interface more clear[0] and "parse-when-needed" solutions less messy.

[0] Example: I'm not sure if advertised service uuids is expected to be in `.details['uuids']` or in `.metadata['uuids']` and both are sort of hidden away IMO.

hbldh, 2020-01-06:
To be honest, I never intended to handle any kind of object for the detected devices. I aimed at making Bleak a tool for talking to devices, not handle the stuff around. I finally added a BLEDevice to store some details that were needed, but never thought that people actually were interested in anything else than seeing the name and getting the address.

Thus, I never created backend specific implementations of the BLEDevice, because I deemed it to be too much of a hassle to handle that. If people wanted to use it they were free to get whatever they wanted from the `details` field and when they wanted to make changes they could stick it the `metadata` dict. They were not inteded for general consumption, but are available for the advanced users.

So please, do not be afraid to modify the BLEDevice object and send a PR with what you propose. Make it work for your specific backend (or all of them I you are capable and want to do it) and I will integrate it. Do the changes in the current BLEDevice and I could possibly break it up later on, when the new scanning interface is done.

PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh