hbldh/bleak #145: bluezdbus: txdbus RemoteError in cleanup

5frank, 2020-01-14:
* bleak: 0.5.1
* os: Linux-4.9.0-11-amd64-x86_64-with-debian-9.11
* python: 3.5.3
* bluez: 5.43

### Description
if notifications are enabled for a device and this device suddenly disconnect, cleanup
code crashes as follows:

```
INFO:bleak.backends.bluezdbus.client:605: GATT Char Properties Changed: gorg/bluez/hci0/dev_F2_1F_2B_52_48_9E/service000e/char0017 | [{'Value': [23, 8, 160]}, []]
DEBUG:bleak.backends.bluezdbus.client:597: DBUS: path: gorg/bluez/hci0/dev_F2_1F_2B_52_48_9E/service000e/char0017, domain: org.bluez.GattCharacteristic1, body: {'Value': [2, 188, 1, 83]}
INFO:bleak.backends.bluezdbus.client:605: GATT Char Properties Changed: gorg/bluez/hci0/dev_F2_1F_2B_52_48_9E/service000e/char0017 | [{'Value': [2, 188, 1, 83]}, []]
DEBUG:bleak.backends.bluezdbus.client:597: DBUS: path: gorg/bluez/hci0/dev_F2_1F_2B_52_48_9E, domain: org.bluez.Device1, body: {'ServicesResolved': False, 'Connected': False}
DEBUG:bleak.backends.bluezdbus.client:622: Device F2:1F:2B:52:48:9E disconnected.
DEBUG:bleak.backends.bluezdbus.client:149: Removing rule PropChanged, ID: 1
DEBUG:bleak.backends.bluezdbus.client:166: Disconnecting from BLE device...
Traceback (most recent call last):
  [...]
  File "/home/sfrank/.local/lib/python3.5/site-packages/bleak/backends/bluezdbus/client.py", line 168, in disconnect
    await self._cleanup()
  File "/home/sfrank/.local/lib/python3.5/site-packages/bleak/backends/bluezdbus/client.py", line 156, in _cleanup
    *(self.stop_notify(_uuid) for _uuid in self._subscriptions)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/home/sfrank/.local/lib/python3.5/site-packages/bleak/backends/bluezdbus/client.py", line 529, in stop_notify
    ).asFuture(self.loop)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
txdbus.error.RemoteError: org.bluez.Error.Failed: No notify session started
```

The [documentation](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt) is not clear to me, but I'm guessing one should not need to "call" `StopNotify` after a disconnect?

btw what is the desirable behavior when a device disconnect? I see some alternatives:

* 1. `stop_notify()` should never raise error due to disconnect disconnect?
* 2. raise specific BleakDisconnectError so it can be distinguished from other errors?
* 3. `_cleanup()` never raises error but stop_notify() will
* 4. something else?

Possible related issue:
* https://github.com/hbldh/bleak/issues/108
* https://github.com/hbldh/bleak/issues/130

5frank, 2020-01-16:
Tested on the following system with same result:
* bleak: 0.5.1
* os: Linux-4.19.0-6-amd64-x86_64-with-debian-10.2
* python: 3.7.3
* bluez: bluetoothctl: 5.50

hbldh, 2020-01-23:
There are multiple scenarios leading to calling the `_cleanup` method, of which abruptly disconnected is one of them. I always want to try to remove all notifications upon disconnect/client removal, so I try to call `StopNotify` on all char uuids registered to have notifications in Bleak. I have no idea how you ended up in your state; I thought it impossible, but apparently not...

IMO:

1. `stop_notify` should always raise an error if the underlying system reports an error. 
2. I could wrap errors form underlying systems, but I don't feel that it will add anything. Furthermore, Bleak has no idea that your device is disconnected, so I cannot reliably raise a Disconnect error...
3. This. This will be implemented. I thought everything error capable had been try-catched in `_cleanup`, but apparently not.

5frank, 2020-01-23:
I think my issue can be reproduced by setting `conn_max_interval` to something low
`echo 24 | sudo dd of=/sys/kernel/debug/bluetooth/hci0/conn_max_interval`. This should cause bluez to disconnect a device after 30 sec or more due to rejected "connection parameter update". I am speculating the reason this error have not been seen before is because it is usually the BLE device (master) and not bluez (client) that causes the disconnect.

hbldh, 2020-01-23:
I made changes in a new commit, f854a24, is the error still present when running it with the Bleak currently in the `develop` branch?

5frank, 2020-01-24:
@hbldh Yay, error is gone! Thanks!
Tested with `commit f123ae0 (HEAD -> develop, origin/develop)`

Log output (if relevant):
```
DEBUG:bleak.backends.bluezdbus.client:190: Disconnecting from BLE device...
DEBUG:bleak.backends.bluezdbus.client:220: Attempt to stop Twisted reactor failed: Can't stop reactor that isn't running.
```

I also noted there is a new extra argument to  the `disconnected_callback` that is not (yet) in the BaseClient.

commit f854a24a (2020-01-23): Fixes #145.
PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh