hbldh/bleak #147: Install requirements.txt

jonattanva, 2020-01-14:
- bleak: 0.5.1
- Windows 10
- python: 3.8.1

### Description

error: option --single-version-externally-managed not recognized

```
Collecting bleak==0.5.1
  Using cached https://files.pythonhosted.org/packages/ce/32/25df87dfa1657c09beb47c49a247f04ee46929ca5be044504615b019a03c/bleak-0.5.1-py2.py3-none-any.whl
Collecting pythonnet; platform_system == "Windows"
  Using cached https://files.pythonhosted.org/packages/4a/0a/5964a5a1fdf207fab8e718edd1f8d3578e0681577fe9abe0d9006f9718c2/pythonnet-2.4.0.tar.gz
Building wheels for collected packages: pythonnet
  Building wheel for pythonnet (setup.py): started
  Building wheel for pythonnet (setup.py): finished with status 'error'
  Running setup.py clean for pythonnet
Failed to build pythonnet
Installing collected packages: pythonnet, bleak
    Running setup.py install for pythonnet: started
    Running setup.py install for pythonnet: finished with status 'error'

  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\jonattan\Documents\aifit\aifit-ble\venv\Scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"'; __file__='"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\jonattan\AppData\Local\Temp\pip-wheel-qeh9kn1n' --python-tag cp38
       cwd: C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\
  Complete output (69 lines):
  running bdist_wheel
  running build
  running build_ext
  Comprobando actualizaciones desde https://www.nuget.org/api/v2/.
  Se est  ejecutando NuGet.exe 4.1.0.
  Actualizando NuGet.exe a 5.4.0.
  Actualizaci¢n correcta.
  Cannot find the specified version of msbuild: '14'
  MSBuild auto-detection: using msbuild version '4.0' from 'C:\Windows\Microsoft.NET\Framework64\v4.0.30319'.
  Restoring NuGet package NUnit.3.7.1.
  Restoring NuGet package UnmanagedExports.1.2.7.
  Restoring NuGet package NUnit.ConsoleRunner.3.7.0.
  Adding package 'NUnit.ConsoleRunner.3.7.0' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
  Adding package 'NUnit.3.7.1' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
  Adding package 'UnmanagedExports.1.2.7' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
  Added package 'UnmanagedExports.1.2.7' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
  Added package 'NUnit.ConsoleRunner.3.7.0' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
  Added package 'NUnit.3.7.1' to folder 'C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\packages'
 
  NuGet Config files used:
      C:\Users\jonattan\AppData\Roaming\NuGet\NuGet.Config
 
  Feeds used:
      C:\Users\jonattan\.nuget\packages\
      https://api.nuget.org/v3/index.json
 
  Installed:
      3 package(s) to packages.config projects
  Traceback (most recent call last):
    File "tools\geninterop\geninterop.py", line 24, in <module>
      from pycparser import c_ast, c_parser
  ModuleNotFoundError: No module named 'pycparser'
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\setup.py", line 623, in <module>
      setup(
    File "C:\Users\jonattan\Documents\aifit\aifit-ble\venv\lib\site-packages\setuptools\__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\setup.py", line 610, in run
      return bdist_wheel.bdist_wheel.run(self)
    File "C:\Users\jonattan\Documents\aifit\aifit-ble\venv\lib\site-packages\wheel\bdist_wheel.py", line 192, in run
      self.run_command('build')
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\command\build_ext.py", line 340, in run
      self.build_extensions()
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\command\build_ext.py", line 449, in build_extensions
      self._build_extensions_serial()
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\distutils\command\build_ext.py", line 474, in _build_extensions_serial
      self.build_extension(ext)
    File "C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\setup.py", line 298, in build_extension
      subprocess.check_call([sys.executable, geninterop, interop_file])
    File "C:\Users\jonattan\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 364, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['C:\\Users\\jonattan\\Documents\\aifit\\aifit-ble\\venv\\Scripts\\python.exe', 'tools\\geninterop\\geninterop.py', 'src\\runtime\\interop38.cs']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for pythonnet
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\jonattan\Documents\aifit\aifit-ble\venv\Scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"'; __file__='"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\jonattan\AppData\Local\Temp\pip-record-vs4igshx\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\jonattan\Documents\aifit\aifit-ble\venv\include\site\python3.8\pythonnet'
         cwd: C:\Users\jonattan\AppData\Local\Temp\pycharm-packaging\pythonnet\
    Complete output (6 lines):
    usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: setup.py --help [cmd1 cmd2 ...]
       or: setup.py --help-commands
       or: setup.py cmd --help
   
    error: option --single-version-externally-managed not recognized
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\jonattan\Documents\aifit\aifit-ble\venv\Scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"'; __file__='"'"'C:\\Users\\jonattan\\AppData\\Local\\Temp\\pycharm-packaging\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\jonattan\AppData\Local\Temp\pip-record-vs4igshx\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\jonattan\Documents\aifit\aifit-ble\venv\include\site\python3.8\pythonnet' Check the logs for full command output.
```

hbldh, 2020-01-23:
The [`pythonnet`](https://github.com/pythonnet/pythonnet) dependency has no built wheels for 3.8 yet, so you have to be able to build it by yourself to run Python 3.8 as of yet. See [here](http://pythonnet.github.io/) for details.

There is also this: https://github.com/pythonnet/pythonnet#python-380-support

Try installing Python 3.7 and using that for Bleak in the meantime.

closed 2020-01-23 by hbldh

hbldh, 2020-06-23:
Use `pythonnet` version 2.5.1 for Python 3.8 support in Windows.

standarddeviant, 2021-04-23:
If anyone is having trouble installing the `pythonnet` dependency on Windows, there are pre-built wheels available on Chris Gohlke's site here: https://www.lfd.uci.edu/~gohlke/pythonlibs/#pythonnet

That site exists to simplify the installation of common python libraries on Windows that typically require a C/C++ compiler to be installed and configured.