hbldh/bleak #149: Connection interval and connection event length change values in bleak

my-abousamra, 2020-01-16:
* bleak version:
0.5.1
* Python version:
3.5.2
* Operating System:
Ubuntu 16.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 
5.43

### Description

I can't find a way to set Min/Max connection interval and Min/Max connection event length in bleak at a connection establishment. Also, I can't find API to make a connection parameter update request to change them during a connection. Is there a plan to add such feature?
Note: I can set Min/Max connection interval from a Linux terminal like (echo 6 | sudo dd of=/sys/kernel/debug/bluetooth/hci0/conn_min_interval) but there's no way to do so for CE length.

5frank, 2020-01-17:
I believe the reason these parameters are not accessible in any of the backend API:s is because they are often decided by the peripheral (i.e. the other side of the communication).
I could not find it in the BLE specs but here are two comments on the subject [1](https://lkml.org/lkml/2019/7/15/981) [2](https://stackoverflow.com/a/45745301).

hbldh, 2020-01-18:
There is no plan to add any such functionality currently. I have not seen mention of such parameters in any of the three different os backends, so I cannot help you there.

my-abousamra, 2020-01-19:
@5frank The central device can also change the min/max connection interval. I saw that in sniffing the connection where the central sent the new values at connection creation and the peripheral accepted them. The problem that the central send min/max CE length with zero value but I want to change them. I thought it may be added to bleak at the starting connection step.

closed 2020-01-19 by my-abousamra

hbldh, 2020-01-23:
I am sorry but I do not know how to change these. This is all I find on this: https://stackoverflow.com/a/45745301

5frank, 2020-01-24:
@my-abousamra Correcting myself: the ["Bluetooth management socket" api](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mgmt-api.txt) do have support for these settings. The only example I can find is in the [source](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/tools/hcitool.c) for the now deprecated `hcitool`. So I guess it theoretically should be possible to do it in python. I believe/hope these settings will be more accessible in future versions of bluez...

ford-prefect, 2024-04-24:
Not sure what your use-case is, but there's a request to do this in BlueZ: https://github.com/bluez/bluez/issues/731