hbldh/bleak #153: Exception on usage of  `await client.write_gatt_descriptor()`

jabezwinston, 2020-01-31:
* bleak version: 0.5.1
* Python version:  3.6.4
* Operating System: Windows 10.0.17763

### Description

I tried to enable notifications by using the below code
```
 await client.write_gatt_descriptor(19,bytearray([0x01,0x00]))
```

and got the exception

```
TypeError: object does not implement IAsyncOperation`1
```

### Crash logs

```
Executing <Handle <TaskWakeupMethWrapper object at 0x0AA966F0>(<Future finis...events.py:275>) created at C:\Python36-32\lib\asyncio\locks.py:249> took 0.125 seconds
Traceback (most recent call last):
  File ".\ota.py", line 133, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Python36-32\lib\asyncio\base_events.py", line 467, in run_until_complete
    return future.result()
  File ".\ota.py", line 77, in run
    await client.write_gatt_descriptor(19,bytearray([0x01,0x00]))
  File "C:\Python36-32\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python36-32\lib\site-packages\bleak\backends\dotnet\client.py", line 421, in write_gatt_descriptor
    descriptor.obj.WriteValueAsync(writer.DetachBuffer())
TypeError: object does not implement IAsyncOperation`1
```

hbldh, 2020-02-03:
The descriptor does not support writing then?

If you want to start notifications on a characteristic, try using the `start_notify` method instead.

jabezwinston, 2020-02-04:
The descriptor supports writing. 
It works with `start_notify`  ! Read back using `read_gatt_descriptor` and obtained `[0x01,0x00]`

closed 2020-03-27 by hbldh