hbldh/bleak #157: Cannot write to a characteristic

Fab64, 2020-02-09:
* bleak version: 0.5.1
* Python version: 3.7.4
* Operating System: Windows 10 /macOS Catalina 

### Description
Windows: I try to write to a characteristic I get the following error

    await client.write_gatt_char(char.uuid, b'Info=100', False)
  File "C:\Users\xxxx\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 401, in write_gatt_char
    data, characteristic.uuid, write_result.Status
bleak.exc.BleakError: Could not write value b'Info=100' to characteristic 00002a00-0000-1000-8000-00805f9b34fb: 3

macOS: call to write_gatt_char never returns

### What I Did

My code:

`
import platform
import asyncio
import logging

from bleak import BleakClient

async def run(address, loop, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        loop.set_debug(True)
        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address, loop=loop) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))

        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))
            for char in service.characteristics:
                if "read" in char.properties:
                    try:
                        value = bytes(await client.read_gatt_char(char.uuid))
                    except Exception as e:
                        value = str(e).encode()
                else:
                    value = None
                log.info(
                    "\t[Characteristic] {0}: ({1}) | Name: {2}, Value: {3} ".format(
                        char.uuid, ",".join(char.properties), char.description, value
                    )
                )
                for descriptor in char.descriptors:
                    value = await client.read_gatt_descriptor(descriptor.handle)
                    #log.info("\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(descriptor.uuid, descriptor.handle, bytes(value)))

                write_value = bytearray([0xa0])
                await client.write_gatt_char(char.uuid, b'Info=100', False)
                log.info('write completed')

if __name__ == "__main__":
    address = (
        "FA:F3:CE:5E:E0:0F"
        if platform.system() != "Darwin"
        else "6CFF7923-7B84-41D3-B021-C823949216C1"
    )
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address, loop, True))
`

5frank, 2020-02-09:
Have you checked that the uuid `00002a00-0000-1000-8000-00805f9b34fb` is writable? A search indicated that this uuid is the "Device Name" which I do not think is a writable characteristic.

Fab64, 2020-02-09:
Thanks for the prompt answer.
My device is a custom one with only one service and one characteristic which is writable. Everything works fine with other applications.

I tried a different code which confirms that the characteristic is writable.

`
    async with BleakClient(address, loop=loop) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))
        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))
            for char in service.characteristics:
                #await client.start_notify(char.uuid, callback)
                if "read" in char.properties:
                    try:
                        value = bytes(await client.read_gatt_char(char.uuid))
                    except Exception as e:
                        value = str(e).encode()
                else:
                    value = None
                log.info(
                    "\t[Characteristic] {0}: ({1}) | Name: {2}, Value: {3} ".format(
                        char.uuid, ",".join(char.properties), char.description, value
                    )
                )
                for descriptor in char.descriptors:
                    value = await client.read_gatt_descriptor(descriptor.handle)
                    log.info(
                        "\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(
                            descriptor.uuid, descriptor.handle, bytes(value)
                        )
                    )

                await asyncio.sleep(15.0, loop=loop)
                # await client.stop_notify(char.uuid)
                log.info("Writing")
                log.info("Writing")
                await client.write_gatt_char(char.uuid, b'Info=100', True)
                #write_value = bytearray([0xa0])
                #await client.write_gatt_char(char.uuid, write_value)
                log.info("Write completed")
                await asyncio.sleep(1.0, loop=loop)
`
This is the output:

Connected: True
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (read,write) | Name: , Value: b'ScaleFab' 
Writing
Writing
Traceback (most recent call last):
  File "C:\Users\Fabrizio\Downloads\Python - BLE\Python - BLE\service_explorer.py", line 77, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "C:\Users\Fabrizio\Downloads\Python - BLE\Python - BLE\service_explorer.py", line 63, in run
    await client.write_gatt_char(char.uuid, b'Info=100', True)
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 401, in write_gatt_char
    data, characteristic.uuid, write_result.Status
bleak.exc.BleakError: Could not write value b'Info=100' to characteristic 00002a00-0000-1000-8000-00805f9b34fb: 3
>>> 

I also noticed another issue on windows only. If I try to set the notification:

await client.start_notify(char.uuid, callback)

I get:

Connected: True
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
Traceback (most recent call last):
  File "C:\Users\Fabrizio\Downloads\Python - BLE\Python - BLE\service_explorer.py", line 77, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "C:\Users\Fabrizio\Downloads\Python - BLE\Python - BLE\service_explorer.py", line 38, in run
    await client.start_notify(char.uuid, callback)
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 459, in start_notify
    status = await self._start_notify(characteristic.obj, callback)
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 518, in _start_notify
    loop=self.loop,
  File "C:\Users\Fabrizio\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\utils.py", line 73, in wrap_IAsyncOperation
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.Exception: The attribute cannot be written. (Exception from HRESULT: 0x80650003)
>>> 

This works on macOS and data are received as expected.

5frank, 2020-02-09:
hmm, not sure what is going on. I now little about the windows backend.
It looks like you already tried to write without response ( i.e. `write_gatt_char(<uuid>, <data> False)`?). is the result/error the same?
Btw. it would help if you [formatted](https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks#fenced-code-blocks)  your code as it is hard to read without indentation.

hbldh, 2020-03-27:
In Bleak version 0.5.1 there is an error using WriteWithResponse. Try installing version 0.6.1 and try again. It might work better. 

The windows issue relates to this: https://stackoverflow.com/questions/48973287/failed-to-subscribe-to-notification-characteristic-in-bluetooth-low-energy
It should not happen. Run the `service_explorer.py` in Windows as well and confirm that the char that you want to start notifications on is actually notifiable.

hbldh, 2020-06-02:
The Core Bluetooth side of this issue is solved in 0.6.4. As for the .NET I have not heard of this error from any other user. Will close for now.

closed 2020-06-02 by hbldh