hbldh/bleak #158: Unexpected errors when attempting to use service_explorer.py

alexc16109935, 2020-02-13:
* bleak version: 0.5.1
* Python version: 3.7.4
* Operating System: Windows 10 Build 18363
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I simply ran the service_explorer.py function after making sure that my board was discoverable by the discover.py function. The GATT attributes failed to print in a variety of ways, and I'm not sure why.

### What I Did

I simply ran service_explorer.py over and over without changing anything.
Note that I was able to get every GATT attribute to print correctly earlier last year, but having changed nothing on my chip, its firmware, or the Bleak code, it's suddenly failing as shown below.

Example 1:
```
Traceback (most recent call last):
  File "service_explorer.py", line 69, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "service_explorer.py", line 29, in run
    async with BleakClient(address, loop=loop) as client:
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\client.py", line 43, in __aenter__
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_compl144, in connectete
    return future.result()                                                                                               244, in get_services
  File "service_explorer.py", line 29, in run
    async with BleakClient(address, loop=loop) as client:                                                                3, in wrap_IAsyncOperation
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\client.py", line 43, in 
__aenter__
    await self.connect()
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 
144, in connect
    services = await self.get_services()
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line ete
244, in get_services
    loop=self.loop,
  File "C:\Users\alchang\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\utils.py", line 73, in wrap_IAsyncOperation                                                                                               __aenter__
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.Exception: Operation aborted (Exception from HRESULT: 0x80004004 (E_ABORT))
```

Example 2:
```
Connected: True
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
        [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (read,notify) | Name: , Value: b'Could not read characteristic value for 00002a05-0000-1000-8000-00805f9b34fb: 2'
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x02\x00'
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
        [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'NXP_WIRELESS_UART'
        [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'\x00\x00'
[Service] 01ff0100-ba5e-f4ee-5ca1-eb1e5e4b1ce0: Unknown
        [Characteristic] 01ff0101-ba5e-f4ee-5ca1-eb1e5e4b1ce0: (write-without-response) | Name: , Value: None
[Service] 0000180f-0000-1000-8000-00805f9b34fb: Battery Service
        [Characteristic] 00002a19-0000-1000-8000-00805f9b34fb: (read,notify) | Name: , Value: b'('
                [Descriptor] 00002904-0000-1000-8000-00805f9b34fb: (Handle: 19) | Value: b"\x04\x00\xad'\x01\x01\x00"    
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 20) | Value: b'\x00\x00'
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
        [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.Exception: Operation aborted (Exception from HRESULT: 0x80004004 (E_ABORT))'
        [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.InvalidOperationException: A method was called at an unexpected time. (Exception from HRESULT: 0x8000000E)'
        [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.ObjectDisposedException: 
The object has been closed. (Exception from HRESULT: 0x80000013)'
        [Characteristic] 00002a27-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.ObjectDisposedException: 
The object has been closed. (Exception from HRESULT: 0x80000013)'
        [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.ObjectDisposedException: 
The object has been closed. (Exception from HRESULT: 0x80000013)'
        [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'System.ObjectDisposedException: 
The object has been closed. (Exception from HRESULT: 0x80000013)'
```

hbldh, 2020-03-27:
I have very little idea as to what causes this. Did you update your Windows installation from last time you ran the scripts? Does it happen reproducibly, in the same manner every time you run it? 

I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to develop branch occasionally, so if you desire these changes you have to do the legwork yourselves.

gyamanishi, 2020-05-05:
I was getting the same error.  I found a pending windows update for Windows 10 to the bluetooth driver.  After I updated, it started to work again.

[image]

closed 2020-06-02 by hbldh

RulerKun, 2020-07-08:
I am also encountering this issue
bleak version: 0.7.1
Python version: 3.7
Operating System: Windows 10 Build 18363.900

Also this is the bluetooth version in my machine
[image]

hbldh, 2020-07-09:
This is an error from the Windows Bluetooth system. I cannot do anything about it I am afraid. If the object has been disposed in the OS code, I cannot handle anything.