hbldh/bleak #159: Not able to connect to device/read services, Windows 10

atc1441, 2020-02-14:
Hi, 

i am using the newest version of bleak and python on windows 10 

I have problems on connecting to BLE devices via bleak but also when using WEB bluetooth with chrome.

When searching for devices they will get shown including their MAC.

I think that this is a bug in the windows 10 ble backend but i dont know where to start to find a solution/answer

it will either error out and say the devices could not be found or it will error with the following:

bleak.exc.BleakError: This characteristic is already present in this BleakGATTServiceCollection!
tested with explore.py example here

even if this does not get answered, maybe other people will have the same problem and can see this

comcute, 2020-03-21:
I can confirm this. Started to happen after with some windows update.

hbldh, 2020-03-27:
This is an unfortunate error that should be looked at asap. It might be due to erronous asumption in the initial architecture of Bleak and be somewhat tricky to address...

I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

hbldh, 2020-06-30:
This should have a solution in version 0.7.0, which is now available on PyPI.

closed 2020-06-30 by hbldh

atc1441, 2020-07-01:
This update seems to have fixed the issue, thanks :) 

But there is another problem in windows 10 witch is not Bleak's fault just want to add it here if someone has the problem as well. 

Windows is not able to connect to devices With HID Characteristics. see also here https://github.com/WebBluetoothCG/registries/issues/24

It is possible that it is caused by the so called Blacklist for WebBluetooth devices, but i am not sure anymore, is Bleak relaying on the WebBluetooth base?

It will then error out with it not being able to get all Characteristics, on devices without HID Chars it does work fine.

hbldh, 2020-07-01:
@atc1441 You're welcome. 

Interesting problem. Please open a new issue with those details so we can try to track that and possibly fix it. But I have to admit, I know next to nothing about WebBluetooth...

thallth, 2021-06-09:
I have the same Issue on Windows 10.
Removing the HID Service fixes it.