hbldh/bleak #160: Add support for async callbacks

5frank, 2020-02-19:
It would be nice if `asyncio.Queue` could be used in notification callbacks. This is currently not possible (if you know a workaround please share!). Perhaps there is also other cases where it is desirable with async callbacks. 
Example: 
```python
import asyncio 

aqueue = asyncio.Queue()

async def run_callback(callback):
    if asyncio.iscoroutinefunction(callback):
        await callback()
    else:
        callback()

def normal_callback():
    print("normal_callback")
    # can not use await here (ofc) 
    # aqueue.put(data) # doesn't work as not awaited

async def async_callback():
    print("async_callback")
    await aqueue.put(data)

async def main():
    await run_callback(normal_callback)
    await run_callback(async_callback)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
```
Do anyone see any problems with adding this feature?

hbldh, 2020-03-27:
Is it even useful? On macOS, I have no control over callbacks, on Windows the callback is [already put on the event loop](https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/client.py#L574), but Linux could be improved in the same way Windows has been in #148.

The Queue seems to be hard to get identical over all backends, and the value added is quite low. Is it mostly to be able to use async functions for callbacks?

5frank, 2020-05-31:
This is probably not worth the effort then. `Queue` should work and it doesn't seem to be any other cases where there is not a workaround.

closed 2020-05-31 by 5frank