hbldh/bleak #161: Error in stopNotify_() method on MacOs

combatistor, 2020-02-20:
* bleak version: 0.5.1
* Python version: 3.7
* Operating System: macOS Mojave 10.14.6
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

Error while calling bleakClient.start_notify() a second time after stopping it (by calling bleakClient.stop_notify)

### What I Did

This is the error I get:
```
File "/Users/utail/PycharmProjects/bleTest/main.py", line 111, in run_notify
    await myBleDevice.bleakClient.start_notify(myCharac.charac.uuid, myBleDevice.notify_handler)
  File "/Users/utail/PycharmProjects/bleTest/venv/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 284, in start_notify
    characteristic.obj, callback
  File "/Users/utail/PycharmProjects/bleTest/venv/lib/python3.7/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 187, in startNotify_cb_
    self._characteristic_notify_status[cUUID] = True
TypeError: 'bool' object does not support item assignment
```

After checking the code I noticed line 199 of PeripheralDelegate.py in `async def stopNotify_()` method you assign directly self._characteristic_notify_status = False 
instead of 
self._characteristic_notify_status[cUUID] = False

after modifying the line everything worked perfectly

5frank, 2020-02-20:
Sorry, I should have made an issue about this. It is fixed by https://github.com/hbldh/bleak/commit/c9edbbe1b0a08b95e8b2c78f53c7d3ab2fd652e7 in develop branch ans should be included in next release.

combatistor, 2020-02-21:
Okay! When will be the next release out then? I was afraid this bug is already documented and will be fixed but I couldn't find any information about it. 
Sorry for the trouble then, I'll wait for the next release

5frank, 2020-03-11:
Fix included in version 0.6.1. Perhaps this issue can be closed!?

closed 2020-03-27 by hbldh