hbldh/bleak #162: Incremental delay in notifications

progerzua, 2020-02-26:
* bleak version: 0.5.1
* Python version: 3.6.8
* Operating System: Ubuntu 18.04.3 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

I`m working with device, that sends me 180 bytes of data every ~150 ms via notification.
This data is somehow processed(it takes less than 10 ms), and sent to another process for vizualization.

I`m doing some specific things to sensor, and can see delay between my input and incoming packets. At the beginning this input minimal, but it increases with time.

After 3 minutes I see 30 seconds delay in incoming data. This delay getting bigger with time.

After sending stop command to my device I can still see, that packets comes to me. Looks like they`re collecting in some queue on my PC, and I`m getting data from this queue.

Can I do smth with that, to decrease this incremental delay?

progerzua, 2020-02-26:
Hallelujah!

You`ve fixed something in last commits, I`ve installed Alpha version from Develop Branch and it works like a charm!

Should I close this issue?

hbldh, 2020-03-27:
Good that it resolved itself without me knowing!

closed 2020-03-27 by hbldh