hbldh/bleak #163: Second example in the README.md hangs

nbenm, 2020-03-06:
* bleak version:0.5.1
* Python version:3.7.6
* Operating System:MacOs Catalina 10.15.3
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

First example in README.md works fine, but second one hangs

### What I Did

I've put the address printed by first example in second script.
But I don't understand which Mac address to use. I've tried the device one, and the UUID, but in both cases script hangs. Nothing happens and there's no output
This device works fine with a proprietary Android app (I have a pin to type).

I have then tried:
discover.py => works fine
get_services.py => hangs with good uuid in the script
service_explorer.py => idem supra

```
python3 test2.py
No output, and no prompt...
```
Any idea?

nbenm, 2020-03-08:
I have tried the following script:

```
# 2E0D0C39-B05F-45C7-82DF-4067CC3129A4: Beurer GL50EVO

import asyncio
from bleak import BleakClient

address = "2E0D0C39-B05F-45C7-82DF-4067CC3129A4"
MODEL_NBR_UUID = "00002a24-0000-1000-8000-00805f9b34fb"

async def run(address, loop):
    print('1',flush=True)
    async with BleakClient(address, loop=loop) as client:
        print('2',flush=True)
        model_number = await client.read_gatt_char(MODEL_NBR_UUID)
        print("Model Number: {0}".format("".join(map(chr, model_number))))

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address, loop))
```
Only first print is executed.
Device is a glucometer.
Can someone help?

nbenm, 2020-03-08:
Here are some new informations:
Trying code with a mac mini mid 2011 seems to work better.

services.py shows:
```
Services: (
    "<CBService: 0x7fa0c7d63c90, isPrimary = NO, UUID = Glucose>",
    "<CBService: 0x7fa0c7f0cec0, isPrimary = NO, UUID = Device Information>",
    "<CBService: 0x7fa0c7d74330, isPrimary = NO, UUID = 00001530-1212-EFDE-1523-785FEABCD123>",
    "<CBService: 0x7fa0c7d74370, isPrimary = NO, UUID = FFF0>"
)
```
But service_explorer.py shows:

```
Connected: True
[Service] 1808: Glucose
	[Characteristic] 2A18: (notify) | Name: , Value: None 
		[Descriptor] 2902: (Handle: 15) | Value: b'\x00\x00' 
	[Characteristic] 2A34: (notify) | Name: , Value: None 
		[Descriptor] 2902: (Handle: 18) | Value: b'\x00\x00' 
2020-03-08 15:38:47.911 Python[60007:4873233] PyObjC: Converting exception to Objective-C:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 246, in peripheral_didUpdateValueForCharacteristic_error_
    "Failed to read characteristic {}: {}".format(cUUID, error)
bleak.exc.BleakError: Failed to read characteristic 2A51: Error Domain=CBErrorDomain Code=0 "Unknown error." UserInfo={NSLocalizedDescription=Unknown error.}
2020-03-08 15:38:47.922 Python[60007:4873233] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'bleak.exc.BleakError'>: Failed to read characteristic 2A51: Error Domain=CBErrorDomain Code=0 "Unknown error." UserInfo={NSLocalizedDescription=Unknown error.}'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff4fbe80db __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff76f28942 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff4fbe8029 -[NSException raise] + 9
	3   _objc.cpython-37m-darwin.so         0x0000000105781a4e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-37m-darwin.so         0x0000000105756e63 method_stub + 5283
	5   _objc.cpython-37m-darwin.so         0x00000001057813e0 ffi_closure_unix64_inner + 720
	6   _objc.cpython-37m-darwin.so         0x00000001057808f6 ffi_closure_unix64 + 70
	7   CoreBluetooth                       0x00007fff4f67dfcd -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 115
	8   CoreBluetooth                       0x00007fff4f679a1f -[CBPeripheral handleMsg:args:] + 297
	9   CoreBluetooth                       0x00007fff4f6744cb -[CBCentralManager handleMsg:args:] + 198
	10  CoreBluetooth                       0x00007fff4f6828ae __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	11  libdispatch.dylib                   0x00007fff77b115fa _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007fff77b09db8 _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007fff77b1e217 _dispatch_queue_serial_drain + 635
	14  libdispatch.dylib                   0x00007fff77b11166 _dispatch_queue_invoke + 373
	15  libdispatch.dylib                   0x00007fff77b15221 _dispatch_main_queue_callback_4CF + 776
	16  CoreFoundation                      0x00007fff4fba0e09 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	17  CoreFoundation                      0x00007fff4fb635da __CFRunLoopRun + 2586
	18  CoreFoundation                      0x00007fff4fb62927 CFRunLoopRunSpecific + 487
	19  Foundation                          0x00007fff51c7e746 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 277
	20  _objc.cpython-37m-darwin.so         0x0000000105780777 ffi_call_unix64 + 79
	21  ???                                 0x000000010829cd70 0x0 + 4431924592
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```
And second example from README.md shows:
```
1
2
Traceback (most recent call last):
  File "./test2.py", line 20, in <module>
    loop.run_until_complete(run(address, loop))
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 583, in run_until_complete
    return future.result()
  File "./test2.py", line 16, in run
    model_number = await client.read_gatt_char(MODEL_NBR_UUID)
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 167, in read_gatt_char
    raise BleakError("Characteristic {} was not found!".format(_uuid))
bleak.exc.BleakError: Characteristic 2A24 was not found!
```

I'm lost!

BlackHawk1912, 2020-03-13:
>  This device works fine with a proprietary Android app (I have a pin to type).

This sounds a little bit like you need to set the security level, right?
If so, Issue #77 should apply to you.

hbldh, 2020-03-27:
I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

hbldh, 2020-09-25:
This was for an old version of Bleak. Will close for now and hope that it is solved in version 0.8.0.

closed 2020-09-25 by hbldh