hbldh/bleak #169: OSError: [Errno 24] Too many open files

farzadpanahi, 2020-03-19:
bleak version: 0.5.1
Python version: 3.8
Operating System: ArchArm Linux
BlueZ version: 5.54

### What I Did
I wrote a script that keeps trying to connect to a BLE device. It runs in an infinite loop forever. I let it run overnight when the BLE device was not in range and obviously it would be failing to connect. In the morning I check the logs and it is failing with `OSError: [Errno 24] Too many open files`.

### Stacktrace 
```
Traceback (most recent call last):
  File "./my-app.py", line 29, in connect_to_device
    async with BleakClient(device.ble_mac, loop=loop) as client:
  File ".../venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 62, in __init__
    p = subprocess.Popen(["bluetoothctl", "--version"], stdout=subprocess.PIPE)
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1603, in _execute_child
    errpipe_read, errpipe_write = os.pipe()
OSError: [Errno 24] Too many open files
```

### Code
```
async def connect_to_device(device, loop):

    while True:
        try:
            logInfo("connecting to {} ...".format(device.ble_mac))
            async with BleakClient(device.ble_mac, loop=loop) as client:
                logInfo("connected to {}".format(device.ble_mac))
                ble = BleDevice(device, client)
                await ble.run()

                while await ble.is_connected():
                    logDebug('waiting ...')
                    await asyncio.sleep(1)
                
                logInfo('disconnected from {}'.format(device.ble_mac))

        except Exception as e:  # TODO: catch specific exception
            logException(e)

def main():
    devices = get_devices(config)

    loop = asyncio.get_event_loop()

    for d in devices:
        loop.create_task(connect_to_device(d, loop))

    loop.run_forever()

if __name__ == "__main__":
    main()

```

dlech, 2020-03-19:
There are quite a fixes in v0.6.1. Have you tried that version?

farzadpanahi, 2020-03-19:
@dlech Actually I am using the latest v0.6.1 release. (Will update the issue description above)

hbldh, 2020-03-27:
No, it does not work. Bleak or txdbus or (more unlikely) the BlueZ DBus API does not close things appropriately and leaks file descriptors. This leads to this problem in you use it continuously. The simpel solution is to not use it continuously.

I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to develop branch occasionally, so if you desire these changes you have to do the legwork yourselves.

hbldh, 2020-07-20:
I think this was already addressed in #154 and released in version 0.6.2. It might be further improved by #236 though.

closed 2020-07-20 by hbldh