hbldh/bleak #170: write works fine with gatttool not with bleak

nbenm, 2020-03-20:
* bleak version: 0.6.1
* Python version: Python 3.7.3
* Operating System: Debian unstable amd64
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Some commands work fine with gatttool, but not with bleak.

### What I Did

```
# -*- coding: utf-8 -*-
import platform
import logging
import asyncio

from bleak import BleakClient

async def run(address, loop, debug=False):

    async with BleakClient(address, loop=loop) as client:
        x = await client.is_connected()
        print("Connected: {0}".format(x))

        write_value = bytearray([0x02,0x00])
#        await client.write_gatt_char("00002902-0000-1000-8000-00805f9b34fb", write_value, response=True)
        await client.write_gatt_descriptor(0x0017, write_value)

        write_value = bytearray([0x01,0x00])
        await client.write_gatt_descriptor(0x0012, write_value)

        write_value = bytearray([0x01,0x00])
        await client.write_gatt_descriptor(0x000f, write_value)

        def my_handler(sender, data):
            print("{0}: {1}".format(sender, data))

        record_access_control_point = "00002a52-0000-1000-8000-00805f9b34fb"
        await client.start_notify(record_access_control_point, my_handler)
        await asyncio.sleep(5.0, loop=loop)
        await client.stop_notify(record_access_control_point)

if __name__ == "__main__":
    address = ("ED:AC:3E:EA:54:FF" if platform.system() != "Darwin" else "C7C493FE-08D2-455D-A31F-039C11C9C9E6")
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address, loop, False))

Connected: True
Traceback (most recent call last):
  File "beurer.py", line 35, in <module>
    loop.run_until_complete(run(address, loop, False))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "beurer.py", line 16, in run
    await client.write_gatt_descriptor(0x0017, write_value)
  File "/home/c0238/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 517, in write_gatt_descriptor
    ).asFuture(self.loop)
txdbus.error.RemoteError: org.bluez.Error.NotPermitted: Write not permitted
```
Here are the gatttoll commands that work:
```
gatttool -b ED:AC:3E:EA:54:FF -t random -i hci0 --char-write-req -a 0x0017 -n 0200; sleep 1
gatttool -b ED:AC:3E:EA:54:FF -t random -i hci0 --char-write-req -a 0x0012 -n 0100; sleep 1
gatttool -b ED:AC:3E:EA:54:FF -t random -i hci0 --char-write-req -a 0x000f -n 0100; sleep 1
gatttool -b ED:AC:3E:EA:54:FF -t random -i hci0 --char-write-req -a 0x0016 -n 0101 --listen
```
It's the first command (handle 0x0017) that doesn't work with bleak.  

I think I'm probably doing something wrong, but I can't find what.  
Thanks for any help.

bardiabarabadi, 2020-03-25:
I have the same problem. I get:

 `object does not implement IAsyncOperation`1`

after calling `await client.write_gatt_descriptor(handle=handle, data=cmd)`

hbldh, 2020-03-27:
I have not tried writing very much to descriptors directly, have only had to deal with characterstics instead. Try using them instead.

`gatttool` uses a completely different API and is not comparable to Bleak.

I will not look into this in the foreseeable future I am afraid.  I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

hbldh, 2020-09-25:
Will close for now. Reopen if this is still a problem.

closed 2020-09-25 by hbldh