hbldh/bleak #171: [Errno 24] Too many open files when discover Continuously

loongWB, 2020-03-23:
* bleak version: 0.6.1
* Python version: 3.7.3
* Operating System: Raspbian GNU/Linux 10 (buster) 【Pizerow】
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.50

### Description

Raspberry Pi's Bluetooth scans continuously and parse the broadcast data from other ble devices, however when the program scans 506 times in loop, it always throws an exception `[Errno 24] Too many open files.` .
Perhaps the error results from:

> DBus connections are not cleaned up after usage. Multiple calls to discover() result eventually in following error.

This bug should be fixed in v0.5.1, however it reappears in v0.6.1.

### What I Did

``` python
async def run():
    count = 0
    while True:
        try:
            devices = await discover()
            for d in devices:
                data_parse(d)
            count += 1
        except:
            traceback.print_exc()
            print('loop count:', count)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    tasks = [
        run_program(),
        asyncio.ensure_future(run()),
    ]
    loop.run_until_complete(asyncio.wait(tasks))
    loop.close()
```

### Traceback

``` python
Traceback (most recent call last):
  File "/home/pi/client/Testdir/Test.py", line 43, in run
  File "/usr/local/lib/python3.7/dist-packages/bleak-0.6.1-py3.7.egg/bleak/backends/bluezdbus/discovery.py", line 140, in discover
twisted.internet.error.ConnectError: An error occurred while connecting: Failed to connect to any bus address. Last error: [Errno 24] Too many open files.
```

dlech, 2020-03-23:
Duplicate of #169?

loongWB, 2020-03-23:
Yes, i think so. On the line 232 of bleak.backends.bluezbus.discovery.py:

``` python
    # Try to disconnect the System Bus.
    try:
        bus.disconnect()
    except Exception as e:
        logger.error("Attempt to disconnect system bus failed: {0}".format(e))

    try:
        reactor.stop()
    except ReactorNotRunning:
        # I think Bleak will always end up here, but I want to call stop just in case...
        pass
```
It seems that the code above does not work.

hbldh, 2020-03-27:
No, it does not work. Bleak or `txdbus` or (more unlikely) the BlueZ DBus API does not close things appropriately and leaks file descriptors. This leads to this problem in you use it continuously. The simpel solution is to not use it continuously.

I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

loongWB, 2020-03-28:
Actually from the issue #143 perhaps there is another solution for #171.
In class `bleak.backends.bluezdbus.discovery.discover`, each discover will create a `Object: reactor`, 

``` python
from twisted.internet.asyncioreactor import AsyncioSelectorReactor
...
reactor = AsyncioSelectorReactor(loop)
 ```
And each reactor will create a `os.Pipe()`, which opens two files to read and write, i think that's the problem. Also i use a simple solution to solve it.

``` python
Old: reactor = AsyncioSelectorReactor(loop)
New: reactor = kwargs['reactor'] if kwargs.get('reactor') else AsyncioSelectorReactor(loop)
 ```

So if we want to discover continuously, we can define a `reactor` outside the `discover`, like this:
``` python
async def run():
    reactor = AsyncioSelectorReactor(loop)
    while True:
        devices = await discover(reactor=reactor)
        for d in devices:
            print(d)
 ```
I have tried this in my Raspberrypi zero w, and it does work.

hbldh, 2020-04-06:
@loongWB  There was an improvement left hanging in PR #154. I merged that into `develop`, so try the version `0.6.2a1` on that branch out and see if it does what you need.

BillMicro, 2020-06-24:
I have been running basically the same discovery loop for 11'435 times now, with no errors, using:
* bleak version: 0.6.4 (master)
* Python version: 3.7.3
* Operating System: Raspbian GNU/Linux 10 (buster), kernel: 4.19.118-v7+
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

hbldh, 2020-06-24:
This does indeed seem to be fixed then. Good to have feedback on this! Thank you. Closing this.

closed 2020-06-24 by hbldh