hbldh/bleak #172: Exception trying to discover services.

whoopsmith, 2020-03-23:
* bleak version: Version 0.6.1
* Python version:  Python 3.6.9
* Operating System: Linux Ubuntu 18.04.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.48

### Description

Trying to learn how to use this project by playing around.
I'm scanning for devices and trying to discover services.

I have combined the discover example with the service_explorer example.  I discover all the devices and then feed each of those to the service_explorer code. 

Sometimes I get a list of services but sometimes get the exception below.  Is this expected?

### What I Did
```
Scanning: ED:E4:C8:1D:86:45 
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/bleak/backends/bluezdbus/client.py", line 144, in connect
    ).asFuture(self.loop)
txdbus.error.RemoteError: org.bluez.Error.Failed: Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./get_whoop.py", line 80, in <module>
    loop.run_until_complete(service_explorer(address, loop, True))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "./get_whoop.py", line 40, in service_explorer
    async with BleakClient(address, loop=loop) as client:
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 129, in throw
    return self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.6/dist-packages/bleak/backends/client.py", line 44, in __aenter__
    await self.connect()
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 129, in throw
    return self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.6/dist-packages/bleak/backends/bluezdbus/client.py", line 146, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.bluez.Error.Failed: Software caused connection abort
```

hbldh, 2020-03-27:
It is not expected and this is the first time I have seen this issue.

I will not look at this in the foreseeable future I am afraid. I am swamped with other (paid) work and cannot guarantee doing anything with this project for quite some time. I will still address PRs to `develop` branch occasionally, so if you desire these changes you have to do the legwork yourselves.

whoopsmith, 2020-03-27:
Great.  Thank you for the response.  I don't mind digging in and figuring out whats going on.  That will be a great way to learn more about how it works.

Now that I know its not expected I'll drill down and see what I can find.  Like you my day job calls me away but the device(s) I'm trying to connect to are part of my day job so I can justify some paid time to investigate.

corvis, 2020-04-14:
Hey @whoopsmith,

Have you solved your issue? It seems I have the similar one - from time to time when I call connect method I got `org.bluez.Error.Failed: Software caused connection abort` error. And this happens really randomly.

I implemented retry logic which tries to reconnect again and again when this happens. Sometimes it connects from the first attempt quickly, sometimes it takes 2-3 attempts and sometimes 7-8. Basically this makes my software pretty unstable and super slow. 

So any insight on the root cause will be highly appertained.

hbldh, 2020-06-02:
There was quite a long discussion about this in #94. See if anything there helps.

[This repo](https://github.com/50ButtonsEach/fliclib-linux-dist/issues/14) laos had issues with this some years ago. The accepted solution seems to be to restart the bluetooth adapter...

RulerKun, 2020-07-09:
Hi @whoopsmith and @corvis did you resolved this issue? I am also experiencing this issue at raspbian and my bluez is already at 5.50

corvis, 2020-08-26:
@RulerKun unfoirtunately no. It still happens randomly, for some people (combination of hardware and software) it happens more often then for others. The recent example is here: https://github.com/corvis/prana_rc/issues/1#issuecomment-680928556 the guy has raspbian and uses embedded rpi adaptor.

hbldh, 2020-09-02:
I have no news for you regarding this. It might be alleviated if #236 is worked on. It might be due to `txdbus` and how it handles the DBus connections.

closed 2022-07-26 by dlech