hbldh/bleak #173: WatchOS <> macOS Bluetooth Stream

fire17, 2020-03-25:
* bleak version: 0.6.1
* Python version: 3.7
* Operating System: OSX 10.15.3 Catalina

### Description:
## Turn macOS into a Peripheral Device, run advertising, and read data from Central (appleWatch)

Hi there everyone! need some ble assistance..

Found an example from Apple to make an iOS app that turns an iPhone to either a Peripheral (believe called GATT Server) or Central (GATT Client); this is used to send data over BLE from one iPhone to another.. [Transferring Data Between BLE Devices](https://developer.apple.com/documentation/corebluetooth/transferring_data_between_bluetooth_low_energy_devices)
Managed to port the Central to an watchOS app! now I can stream any data I wish from my appleWatch to the iPhone (Peripheral) with super nice fps with really minimal latency at good range. Awesome

NOW trying to get this data to the computer...
Since it's my mother tongue.. i'd really prefer reading this data using python
And hopefully cross-os also, but first macOS..

So.. turning this mac into a peripheral
## What I Found Out
### Don't Work with macOS:
#### bluepy, BlueZ
### Managed to scan for devices:
#### python-bleson, bleak
### Extra
there's PyGATT but looks old af
(Also there's a bleno with NodeJS, managed to install it but could't read data) 

## Bleak
I've found out that there's [PeripheralDelegate.py](https://github.com/hbldh/bleak/blob/master/bleak/backends/corebluetooth/PeripheralDelegate.py) but couldn't find any example usage...
I'll try to hack it abit and see if something works.. Hope im aiming right

If anyone managed to turn **a mac** into **a peripheral** and **read data over ble** by any means or know of any other libraries (python or other) please send a hint!!! very much apreciated!
#### Thanks a lot and have a good one!
#### Tami

hbldh, 2020-03-27:
Do not use the `PeripheralDelegate.py` class directly. Look at the examples in the README.md, the RTD documentation or the examples in [examples folder.](https://github.com/hbldh/bleak/tree/master/examples)

hbldh, 2020-06-02:
This is more of a question on how to use the package, not an actual issue. Will close.

closed 2020-06-02 by hbldh