hbldh/bleak #175: win10 local_name is 'Unknown'

vjani68, 2020-04-07:
* bleak version: 0.6.1
* Python version: 3.6
* Operating System: Win10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I use scanner get_discovered_devices but I see device name's 'Unknown' always

### What I Did

I saw scanner.py in the 135. line:            
 if (
                not new_device.name
                and event_args.BluetoothAddress in self._scan_responses
            ):
                new_device.name = self._scan_responses[
                    event_args.BluetoothAddress
                ].Advertisement.LocalName
            found.append(new_device)
this new_device.name never nothing because in the 160. line:
return BLEDevice(
            bdaddr, local_name, event_args, uuids=uuids, manufacturer_data=data
BLEDevice init local_name = 'Unknown

dlech, 2020-04-07:
I haven't had time to dig into this, but I noticed recently (using Qt bluetooth fwiw) that on Windows, the advertising data and scan response are separated. In other words, there are multiple advertisements that need to be put together in order to get the full response. See [ BluetoothLEAdvertisementType](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementtype).

For example, on the devices I have the list of services and manufacturer specific data are in the ` ConnectableUndirected ` (`ADV_IND`) packet and the local name is in the ` ScanResponse ` (`SCAN_RSP`) packet.

I think bleak needs to be fixed to put the information these two packets (and probably other advertisement types for other devices too) together in a single "response" so that windows behavior matches other platforms.

vjani68, 2020-04-08:
I am using WX. If I use discover than ok. 
the problem is that this line never run
**new_device.name = self._scan_responses[
event_args.BluetoothAddress
].Advertisement.LocalName
found.append(new_device)**

**if (
not new_device.name
and event_args.BluetoothAddress in self._scan_responses
):**
If i change this
if (
new_device.name
and event_args.BluetoothAddress in self._scan_responses
):
then ok

hbldh, 2020-04-21:
I see the problem. I will try to fix so that it does not say `Unknown` before testing if it is empty or None.

pjbosco, 2021-02-21:
This requires an update to the constructor of BLEDevice. If the 'name' parameter is None, the BLEDevice.name gets set to 'Unknown'. 

So long as BLEDevice never accepts None as a name, the code in the following branch in the dotnet scanner.py will never execute:
```
            if (
                not new_device.name
                and event_args.BluetoothAddress in self._scan_responses
            ):
                new_device.name = self._scan_responses[
                    event_args.BluetoothAddress
                ].Advertisement.LocalName
```

Allowing BLEDevice to accept None as a name, and then replacing the code in the dotnet scanner.py with the following fixes the issue for me:
```
            if not new_device.name:
                if event_args.BluetoothAddress in self._scan_responses:
                    new_device.name = self._scan_responses[
                        event_args.BluetoothAddress
                    ].Advertisement.LocalName or "Unknown"
                else:
                    new_device.name = "Unknown"
```

@dlech is right... This of course is contingent upon a ScanResponse type advertisement being available for all device addresses that were found since that is what contains the local name. I have situations where sometimes a device will show Unknown and sometimes it will show its actual name. I think this has more to do with the advertising interval and scans not waiting for a corresponding ScanResponse for each address that has been encountered. 

Hope this helps!

dlech, 2021-02-22:
FYI, I wrote some thoughts about how to make Windows more like other platforms by internally combining the different advertisement types in #394.

hbldh, 2021-02-25:
Yes, I remember that issue. Will close this one in favour of that one, continuing this work.

closed 2021-02-25 by hbldh