hbldh/bleak #177: Wait response flag inversion

nezval, 2020-04-11:
* bleak version: 0.6.1
* Python version:3.6.3
* Operating System: macOS 10.15.4
Was connected to "lego control plus"

### Description

inside of BleakClientCoreBluetooth.write_gatt_char 
line: 
```
success = await cbapp.central_manager_delegate.connected_peripheral_delegate.writeCharacteristic_value_type_(
            characteristic.obj,
            value,
            CBCharacteristicWriteWithResponse if response else CBCharacteristicWriteWithoutResponse
        )
```
where 
CBCharacteristicWriteWithResponse equal 0
CBCharacteristicWriteWithoutResponse equal 1

and inside of writeCharacteristic_value_type_

```
        if response:
            await event.wait()
```
So when write_gatt_char called with `response=False` inside of `writeCharacteristic_value_type_` it will wait forever.

It looks like doing this conversion value right on the call of `writeValue_forCharacteristic_type_` will fix it.

amacf, 2020-04-17:
I also ran into this same issue:

bleak version: 0.6.1
Python version:3.7.4
Operating System: macOS 10.14.4

hbldh, 2020-04-21:
@nezval Ok, this is not one of my finest coding moments. I imported `CBCharacteristicWriteWithResponse` from Core Bluetooth and - without checking - expected it to be equal to 1 and `CBCharacteristicWriteWithoutResponse` equal to 0. Not particularly proud of this, but it is what it is. 

Can you of @amacf install the attempted fix in the develop branch and  see if it alleviates the problems?

amacf, 2020-04-24:
I can confirm version 0.6.2a2 on the develop branch fixed the problem for me!

hbldh, 2020-05-05:
@amacf Very good. i will try to release a 0.6.2 as soon as I can.

PR #193 (2020-05-15): Version 0.6.2
* Better cleanup of Bluez notifications (merges #154 and fixes #130 )
* Fix for ``read_gatt_char`` in Core Bluetooth (fixes #177)
* Fix for ``is_disconnected`` in Core Bluetooth (merges #187 and fixes #185)
* Added ``disconnection_callback`` functionality for Core Bluetooth (merges #186 and fixes #184)
* Documentation fixes
* Added ``requirements.txt``
commit ac3dddd2 (2020-05-15): Merge pull request #193 from hbldh/release/0.6.2

Version 0.6.2

Better cleanup of Bluez notifications (merges #154 and fixes #130 )
Fix for read_gatt_char in Core Bluetooth (fixes #177)
Fix for is_disconnected in Core Bluetooth (merges #187 and fixes #185)
Added disconnection_callback functionality for Core Bluetooth (merges #186 and fixes #184)
Documentation fixes
Added requirements.txt
closed 2020-05-15 by hbldh