hbldh/bleak #178: Operation already in progress

LeonidasFettikus, 2020-04-13:
Hi,

I am using a custom build of bleak with the fix for another problem outlined [here](https://github.com/hbldh/bleak/issues/171). The program I wrote should monitor for specific bluetooth devices indefinitely and although it works, I get lots of errors related to bluez:

```
2020-04-13 01:58:19,603 ERROR    org.bluez.Error.InProgress: Operation already in progress
Traceback (most recent call last):
  File "bleservice.py", line 40, in run
  File "asyncio/base_events.py", line 584, in run_until_complete
  File "bleclient.py", line 26, in run
  File "bleclient.py", line 52, in scan_devices
  File "bleak-0.6.1-py3.7.egg/bleak/backends/bluezdbus/discovery.py", line 204, in discover
txdbus.error.RemoteError: org.bluez.Error.InProgress: Operation already in progress
```
I can handle these errors (unlike the 'too many open files' one, where I had to restart the program). But I need to restart bluetooth adapter every time. I can also do this from code but it's rather ugly and I would like to not do it. I am a total Python noob, so can you advise me on what to do here or is the error related to the bluez library?

Here's my code:

```
import time
import value
import asyncio
import logger
from concurrent.futures import ProcessPoolExecutor
from bleak import discover, BleakClient

class BleClient:
    subscribers: set
    ble_device_wildcard = ''
    uuid_button_pressed = ''
    values: list
    logger: logger.Logger

    def __init__(self, ble_device_wildcard, uuid_button_pressed, logger):
        self.subscribers = set()
        self.ble_device_wildcard = ble_device_wildcard
        self.uuid_button_pressed = uuid_button_pressed
        self.values = list()
        self.logger = logger

    def register_subscriber(self, subscriber):
        self.subscribers.add(subscriber)

    async def run(self, loop, reactor):
        scanned_devices = await self.scan_devices(reactor)
        self.logger.log_information('Finished scanning. Found ' + str(len(scanned_devices)) + ' devices.')
        tasks = list()
        
        for scanned_device in scanned_devices:
            tasks.append(asyncio.create_task(self.get_device_characteristic(scanned_device.address, loop)))

        if(len(tasks) > 0):
            await asyncio.wait(tasks)
            self.send_values()

    async def get_device_characteristic(self, address, loop):
        async with BleakClient(address, loop=loop) as client:
            await client.is_connected()
            self.logger.log_information('Connected to device with address ' + address)
            await client.start_notify(self.uuid_button_pressed, self.notification_handler)
            time.sleep(2)
            await client.stop_notify(self.uuid_button_pressed)
            await client.write_gatt_char(self.uuid_button_pressed, bytearray('1', 'utf-8'))

    def send_values(self):
        for subscriber in self.subscribers:
            subscriber.receive_values(self.values)
        self.values.clear()

    async def scan_devices(self, reactor):
        available_devices = await discover(reactor=reactor)
        devices = list()
        for available_device in available_devices:
            if self.ble_device_wildcard in available_device.name:
                devices.append(available_device)
        return devices

    def notification_handler(self, sender, data):
        strValue = data.decode('utf-8')
        objVal = value.Value(strValue.split(';')[0], strValue.split(';')[1])
        if(objVal.button_press_count > 0 and
        not any(v.button_id == objVal.button_id and v.button_press_count == objVal.button_press_count for v in self.values)):
            self.values.append(objVal)
        self.logger.log_information('Received values from device')
```

I run this from my service class like this:

```
def run(self):
        self.ble_client.register_subscriber(self)
        loop = asyncio.get_event_loop()
        reactor = AsyncioSelectorReactor(loop)

        while True:
            try:
                loop.run_until_complete(self.ble_client.run(loop, reactor))

            except Exception as ex:
                print(ex)
                self.logger.log_exception(ex)
                self.logger.log_error("Error while scanning for ble devices! Trying to reboot bluetooth.")
                os.system('rfkill block bluetooth')
                time.sleep(2)
                os.system('rfkill unblock bluetooth')

            time.sleep(self.config.ble_scan_interval)
```

LeonidasFettikus, 2020-04-13:
Ok, I would like to send a little update on this issue. I have now figured out how to `pip install` directly from a branch, using your `develop` branch as a source. Before, I cloned the repo, switched to the `develop` branch, created an EGG and installed that with `easy_install`. For some reason, that resulted in numerous errors while running `pip install` directly on the `develop` branch did not. I have now run the program for about 45 minutes without any errors popping up, which more than tripled the last record of around 15 minutes of no errors.

hbldh, 2020-04-21:
Whn you do a `async with BleakClient(address, loop=loop) as client:` the `connect` method does a `dsicovery` by iteself in the backgroudn, this is probably what collides with your other code. So the
```python
for scanned_device in scanned_devices:
    tasks.append(asyncio.create_task(self.get_device_characteristic(scanned_device.address, loop)))
```
might lead to multiple calls to dsicovery colliding if there are many devices to connect to. This might change in future versions of bleak, but right now it is how it goes. Try changing to

```python
async with BleakClient(address, timeout=0.0, loop=loop) as client:
```
inside your `get_device_characteristic` method and see if it improves anything.

hbldh, 2020-06-02:
This does not seem to be a bleak bug, but rather an incorrect usage pattern. Closing this.

closed 2020-06-02 by hbldh