hbldh/bleak #179: Bug? Hang when the BLE sensor loses power during a GATT connection.

gyamanishi, 2020-05-02:
* bleak version:0.6.1
* Python version:3.7
* Operating System:Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

First of all, I love the package.  It works.

I've implemented an BLE OTA server on Windows 10.  It is file transfer to a BLE sensor.  All works fine execpt if the sensor goes away in the middle of the GATT connection.  When this happens the python script hangs waiting for a notify (GATT response).  

I tried to implement a disconnect callback and it seems not implemented on .net.

I'm looking for a timeout, or an watchdog behavior when it hangs.  Any suggestions.
```

hbldh, 2020-05-05:
I would like to give you a `disconnect` callback in Windows, but it is only available in the Linux backend right now and I have not time implementing it for Windows right now. This might be impacted by #180 as well, so I have to wait until I can evaluate that first...

hbldh, 2020-09-25:
This is probably solved in version 0.8.0. Disconnect callbacks are also available in Windows in that version. Will close for now.

closed 2020-09-25 by hbldh