hbldh/bleak #180: Change from pythonnet to WinRT to support Python 3.7 and 3.8 on Windows

cHemingway, 2020-05-05:
* bleak version: 0.6.1
* Python version: 3.8
* Operating System: Windows 10 1909
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description
Bleak fails to install on Windows 10 with Python 3.8, as Python 3.8 is not supported by pythonnet, which appears to not officially support python 3.7 or 3.8. 

Microsoft have now released [WinRT](https://pypi.org/project/winrt/), which is an official python bridge to the WindowsRT APIs, and so it might be possible to use this instead of PythonNet for access to Bluetooth on windows. See https://github.com/beeware/toga/issues/884 for another project that is discussing making this change. 

Hopefully, this might solve other open issues tagged with .NET too.

cHemingway, 2020-05-05:
Had a quick go, some immediate issues.

- A lot of base types are missing as `System` namespace doesn't seem to be exposed, so `from System import Array, Byte, UInt64` does not work as `from winrt.system import Array,Byte,UInt64`. The WinRT bridge might convert some of these types automatically, so we don't need to use them directly.

- Enum definitions are different as WinRT capitalises them, so `BluetoothLEAdvertisementType.ScanResponse` becomes `BluetoothLEAdvertisementType.SCAN_RESPONSE`

- `utils.py` uses `System.Threading.Task` which is not exposed. Would need to be replaced with `Windows.System.Threading`

hbldh, 2020-05-06:
This is indeed interesting, especially if the BeeWare project deems it to be. 

Still, it does not seem to solve the Python 3.8 problem in Windows though according to the Toga , so changing is not really worthwhile to do right now.

claybrooks, 2020-06-30:
For what it's worth, I was able to get pythonnet working with python3.8.  `pip install pythonnet` did not work, I had to build from source from their github.

hbldh, 2020-08-19:
Finally got some time to wxplored this path in the `feature/winrt` branch. I have converted the entire `pythonnet` solution to `winrt` and it works, with some caveats.
Apparently `winrt` does not convert `GUID` objects properly, so no `uuid` attributes can be read (raises a `NotImplementedError`).  One can get around this by some ugly regex-solutions for the service uuids, but characteristics has to be queried by handle (or by the characteristic object itself).

It is seemingly slower than the `pythonnet` solution, at least in getting services, characteristics and descriptors. It yielded nicer code and provides a MUCH better way of integrating with the WinRT namespaces. It has only been tested in Win 10 1909 with Python 3.7. 

Sadly, unless the GUID problem can be solved it is however not an option to switch to this I am afraid. I plan to merge it into the codebase and keep it as a extras install so that people who accept the handle approach can use it if it solves e.g. #255. 

The GUID problem is reported in this issue: https://github.com/microsoft/xlang/issues/662

dhalbert, 2020-10-14:
pythonnet 2.5.x (2.5.1 is latest as of now) supports python 3.8.

dlech, 2020-12-06:
FYI, `winrt winrt 1.0.20330.1 ` is now available for Python 3.7, 3.8, and 3.9 and the GUID issue has been fixed.

hbldh, 2020-12-07:
You did find someone to merge your PR finally? Congratulations! I will try it out and merge it eventually.

hbldh, 2021-02-19:
Given that the `pythonnet` project seems to be a bit late with supporting new Python versions, with 3.9 missing a binary wheel for quite some time and 3.10 is well under way I think it is prudent to look at the WinRT backend once again. Since @dlech has fixed the GUID problem and they have released a new version with binary wheels for 3.7, 3.8 and 3.9, it is now seems to be a better option than `pythonnet`.

I did some work on the `feature/winrt` branch that I now moved over to `feature/winrt2` (there were so many Git conflicts that I started a new branch instead...) and brought it up to current `develop` level. I have run it for some time today and it seems to work fine.

It is still slower than the `pythonnet` backend.

I have made WinRT the default requirement in Windows for Python 3.9+ in that branch now (I have not tested it yet with a 3.9 installation through...). It does add another backend to support though. I would vote for moving all Python 3.7+ installations to the WinRT backend as well, maybe with `pythonnet` as an extras (`pip install bleak[pythonnet]`).

I would very much appreciate opinions on this.

dlech, 2021-02-19:
> `pip install bleak[pythonnet]`

I wonder if there is a a way to make both `pythonnet` and `winrt` optional (`extras_require`) and hack `setup.py` so that if one of the two is not specified, then select the best one based the Python runtime version.

I wonder what makes `winrt` slower. Any theories?

hbldh, 2021-02-19:
I would like that also. The most important thing is that if you install bleak with no extras, you get something that works on your system.

#Regarding the speed, if I have the asyncio logger running then each async call in the `get_services` part takes a lot of time. I have no code evidence to back it up though, but it seems that all .Net async calls are really inefficient.

closed 2021-10-05 by dlech