hbldh/bleak #181: Positive RSSI Values ?

seppap, 2020-05-07:
* bleak version: 0.6.1
* Python version: 3.6.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

Reading the RSSI of a Device using the discover() function

Occasionally get a positive RSSI (79 or 80) while typically its been -50.  Want to ensure its real and not a bug.

import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d.name,d.address,d.rssi)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
Output
Unknown XX:XX:XX:XX:XX:XX 79

Have seen it on more than one device.

hbldh, 2020-05-12:
This is strange. I have no idea why. The RSSI is fetched directly from the .NET backend. Maybe bleak needs to cache the RSSI value and not let the connection object persist and fetch it from that. Will look into this eventually.

hbldh, 2020-09-25:
I cannot reproduce this. Will close for now.

closed 2020-09-25 by hbldh

seppap, 2020-09-25:
Appreciate attempting to look into it.  It’s rare even for us. - Paul

[External Email]