hbldh/bleak #182: Targeted discover() by MAC

seppap, 2020-05-07:
* bleak version: 0.6.1
* Python version: 3.6.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

Can discover() support a targeted BLEMAC call

Rather than having the discover() function find all BLE devices, can discover() or another function just attempt to 'discover' one BLEMAC address ?  Working in an environment with > 10 or more BLE devices and it would speed things along.  Or, can discover() stop when it finds a BLEMAC?

Current Code
import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d.name,d.address,d.rssi)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
#
Output
Unknown XX:XX:XX:XX:XX:XX -56
Unknown XX:XX:XX:XX:XX:XX -60
Unknown XX:XX:XX:XX:XX:XX -70
Unknown XX:XX:XX:XX:XX:XX -52
etc

hbldh, 2020-05-12:
You should use the new [BleakScanner](https://github.com/hbldh/bleak/blob/master/bleak/backends/scanner.py#L9) interface then. Do something like
```python
import asyncio
from bleak import BleakScanner

async def run():
    # Register a register_detection_callback here and use a asyncio.Event to
    # break the scanning upon detection of what you seek
    await scanner.start()
    async for _ in range(20):  # Wait 2 seconds
        await asyncio.sleep(0.1, loop=loop)
        # Break here if your desired  
    await scanner.stop()
    devices = await scanner.get_discovered_devices()
    for d in devices:

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

You can make use of the `register_detection_callback` or the `set_scanning_filter` methods; see https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementwatcher.advertisementfilter?view=winrt-18362 and https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/scanner.py#L123-L129 for details about advertisment filters in Windows. Feel free to make PRs if you find a better way of handling it.

Try it out and see if you can achieve what you want.

closed 2020-05-12 by hbldh

seppap, 2020-05-13:
We tried it and it worked well.  Thank You