hbldh/bleak #183: BLE Notifications not working in 0.6.0 and 0.6.1 on Windows

benfwalker, 2020-05-10:
* bleak version: 0.6.1
* Python version: 3.8.2
* Operating System: Windows 10 Build 18363
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

Trying to get notifications from a custom BLE device, but the callback function doesnt run until await client.read_gatt_char is run. Behaviour is consistent on multiple computers and BLE versions (4.2 and 5.0). Notifications are fine on 0.5.1, but dont work on 0.6.0 and 0.6.1

 Output is as expected on bleak 0.5.1:
```
data before notify:  bytearray(b'1#\xed[')
None
Start delay
INTTERUPT data= bytearray(b'\n(.\xfc')
End Delay
Data from interrupt:  bytearray(b'\n(.\xfc')
Data from direct:  bytearray(b'\n(.\xfc')

```
 but on bleak 0.6.0 and 0.6.1:
```
data before notify:  bytearray(b'1#\xed[')
None
Start delay
End Delay
INTTERUPT data= bytearray(b'\n(.\xfc')
Data from interrupt:  bytearray(b'\n(.\xfc')
Data from direct:  bytearray(b'\n(.\xfc')
```

The callback should be running during the time.sleep(5)

### What I Did

```
import platform
import asyncio
import logging
import time

from bleak import BleakClient

DEVICE_REG_CHARACTERISTIC = "f3641401-00b0-4240-ba50-05ca45bf8abc"
DEVICE_VAL_CHARACTERISTIC = "f3641402-00b0-4240-ba50-05ca45bf8abc"

newdata=0
newdataout=0

def modify_command(command, recordnumber):
     temph = int(command, 16) + recordnumber
     return temph.to_bytes(4, byteorder="little")
	
def callback(sender, data):
     global newdata
     global newdataout
     print("INTTERUPT data=",data)
     newdata=1
     newdataout=data

async def run(address, loop, debug=False):
     async with BleakClient(address, loop=loop) as client:
          x = await client.is_connected()
          time.sleep(1)

          print("data before notify: ", await client.read_gatt_char(DEVICE_VAL_CHARACTERISTIC))
		
          print(await client.start_notify(DEVICE_VAL_CHARACTERISTIC, callback))	#Notification start
          time.sleep(1)
          command = modify_command("0x01040000", 5)
          await client.write_gatt_char(DEVICE_REG_CHARACTERISTIC, command)	#This should make the data change
          #this loop is how the code will eventually work. This works as expected on 0.5.1, but on 0.6.0 and 0.6.1, it just hangs forever
         #while newdata == 0:
             #pass
          #newdata=0
          print("Start delay")
          time.sleep(5) #notify should happen here
          print("End Delay")
          newdataouttemp = await client.read_gatt_char(DEVICE_VAL_CHARACTERISTIC)
          print("Data from interrupt: ",newdataout)
          print("Data from direct: ", newdataouttemp)
		

if __name__ == "__main__":
     address = "##:##:##:##:##:##" #Censored 
     loop = asyncio.get_event_loop()
     loop.run_until_complete(run(address, loop, True))
```
[Test1temp.txt](https://github.com/hbldh/bleak/files/4605568/Test1temp.txt)

benfwalker, 2020-05-10:
This appears to be happening because of the change made in 84d923d. Having everything be on the same thread means that 'time.sleep(5)' needs to be `await asyncio.sleep(5.0)`. I'm not fully versed on the intricacies of the async library, but if possible, it might be good to have notifications run on a separate thread.

hbldh, 2020-05-12:
That is right. Never use `time.sleep` when using `asyncio`.

No, the callbacks happen exactly where they should. They are put on the event loop end executed when they are properly scheduled. The pre 84d923d situation was that the callback function was run when the dotnet callback arrived, now it is scheduled on the event loop instead, which is better.

closed 2020-05-12 by hbldh