hbldh/bleak #184: Feature request: Add disconnection callback in CoreBluetooth

hardik-prajapati, 2020-05-12:
* bleak version: 0.6.1
* Python version: 3.7.4
* Operating System: Mac OS Mojave
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description
Disconnection callback not implemented in CoreBluetooth

### What I Did
```
client.set_disconnected_callback(disconnection_callback)
```

From the code it seems we can call callback function from the function  `centralManager_didDisconnectPeripheral_error_` from `CentralManagerDelegate.py`.

I can do that raise PR for it if you want.

hbldh, 2020-05-12:
No, it is not implemented. Feel free to do it if you want it.

bsiever, 2020-05-13:
For what it's worth,
-  I've implemented this (disconnect callbacks of macOS)
- fixed `is_connected` on macOS
- added support for multiple clients on macOS
- partially implemented scanning on macOS (need to test and add support for filters)

I'm going to do a bit more testing and try to issue a pull request in the coming weeks.  In the meantime, here's my fork/branch: https://github.com/bsiever/bleak/tree/macOS-multiple-clients

(And it is a great library --- thanks!)

hardik-prajapati, 2020-05-13:
Created pull request here : https://github.com/hbldh/bleak/pull/186

PR #193 (2020-05-15): Version 0.6.2
* Better cleanup of Bluez notifications (merges #154 and fixes #130 )
* Fix for ``read_gatt_char`` in Core Bluetooth (fixes #177)
* Fix for ``is_disconnected`` in Core Bluetooth (merges #187 and fixes #185)
* Added ``disconnection_callback`` functionality for Core Bluetooth (merges #186 and fixes #184)
* Documentation fixes
* Added ``requirements.txt``
commit ac3dddd2 (2020-05-15): Merge pull request #193 from hbldh/release/0.6.2

Version 0.6.2

Better cleanup of Bluez notifications (merges #154 and fixes #130 )
Fix for read_gatt_char in Core Bluetooth (fixes #177)
Fix for is_disconnected in Core Bluetooth (merges #187 and fixes #185)
Added disconnection_callback functionality for Core Bluetooth (merges #186 and fixes #184)
Documentation fixes
Added requirements.txt
closed 2020-05-15 by hbldh