hbldh/bleak #185: client.is_connected returns `True` even if the device is disconnected.

hardik-prajapati, 2020-05-12:
* bleak version: 0.6.1
* Python version: 3.7.4
* Operating System: Mas OS Mojave
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

Device connected state is not properly maintained. 
```
connected = await self._client.is_connected()
```
`is_connected` returns `True` even if the device is disconnected.

Below are the logs:
```
connected: True
connected: True
2020-05-12 16:12:58,154 - bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG - Peripheral Device disconnected!
connected: True
connected: True
```

### Solution
bleak/backends/corebluetooth/CentralManagerDelegate.py :
```
    def isConnected(self) -> bool:
        # Validate this
        return self.connected_peripheral != None
```
I think you should return `self._connection_state == CMDConnectionState.CONNECTED` instead of `self.connected_peripheral != None`

I can raise PR for this if you want.

Thanks, for the great library :) It is really useful.

dlech, 2020-05-12:
Yes, please make a pull request if it fixes the problem.

hbldh, 2020-05-12:
This is a valid point. Please make that PR.

hardik-prajapati, 2020-05-13:
Created pull request here : https://github.com/hbldh/bleak/pull/187

hbldh, 2020-05-13:
Merged the PR. Will close this when I make a new release. Thank you!

PR #193 (2020-05-15): Version 0.6.2
* Better cleanup of Bluez notifications (merges #154 and fixes #130 )
* Fix for ``read_gatt_char`` in Core Bluetooth (fixes #177)
* Fix for ``is_disconnected`` in Core Bluetooth (merges #187 and fixes #185)
* Added ``disconnection_callback`` functionality for Core Bluetooth (merges #186 and fixes #184)
* Documentation fixes
* Added ``requirements.txt``
commit ac3dddd2 (2020-05-15): Merge pull request #193 from hbldh/release/0.6.2

Version 0.6.2

Better cleanup of Bluez notifications (merges #154 and fixes #130 )
Fix for read_gatt_char in Core Bluetooth (fixes #177)
Fix for is_disconnected in Core Bluetooth (merges #187 and fixes #185)
Added disconnection_callback functionality for Core Bluetooth (merges #186 and fixes #184)
Documentation fixes
Added requirements.txt
closed 2020-05-15 by hbldh