hbldh/bleak #188: Can not install bleak on Ubuntu / Win10

sonus89, 2020-05-13:
* bleak version: 0.6.1
* Python version: 3.8.2
* Operating System: Ubuntu / Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

I get this error message after executing 
sudo pip3 install bleak
I also tried to install setuptools but no effect ...

Building wheel for twisted (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-fg8z9km8/twisted/setup.py'"'"'; __file__='"'"'/tmp/pip-install-fg8z9km8/twisted/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-qzdnck2u

```

hbldh, 2020-05-13:
On Windows you cannot use Python 3.8 due to the required package `pythonnet` not having binary wheels for 3.8.

As for Ubuntu, I was not aware of any problem prior to this with 3.8. Could you please format your traceback better so I can understand what the acual error is? The text posted above is not the entire error, right?

Also, never instal python packages with sudo. Create a virtualenv and install into that instead. Never, ever install python packages in the system python on Ubuntu.

sonus89, 2020-05-13:
Thank you for the answer!
1. What is the latest compatible Python version for bleak on Windows?
2. How do I properly install bleak on linux? ( First I tried to add the module in PyCharm )

hbldh, 2020-05-13:
It works with 3.5, 3.6 and 3.7 (see [setup.py](https://github.com/hbldh/bleak/blob/master/setup.py#L106-L110)).

You install in via pip/pip3. But not with sudo, but in a virtualenv. Read [Brett Cannon's primer](https://snarky.ca/why-you-should-use-python-m-pip/) on `pip` to understand why.

hbldh, 2020-06-02:
Note to self: Better documentation about Python 3.8 not being supported in Windows needs to be written.

hbldh, 2020-06-24:
Install `pythonnet` >= 2.5.1 in Windows 10. This requirement will be updated in bleak 0.7.0

closed 2020-06-24 by hbldh