hbldh/bleak #189: Can't read characteristic longer than 20 bytes.

gyamanishi, 2020-05-13:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I am trying to read a 200 byte characteristic, but the read receives 20 bytes.  The documentation on reading a blob (large characteristic)  is not clear.

### What I Did
Verified the MTU exchange happened.  I can write a 244 byte characteristic to a TI launchpad.
When I issue a read, it only receives 20 bytes.

I can't find where you specify the length of the read, or the offset.  The microsoft doc mentions an error code that is the characteristic is not long, hinting at the fact that it reads blobs.
```
Paste the command(s) you ran and the output.
Connected: True
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (write-without-response,write) | Name: , Value: None 
	[Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (write-without-response,write) | Name: , Value: None 
	[Characteristic] 00002a04-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'P\x00\xa0\x00\x00\x00\xe8\x03' 
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
[Service] f000ffd0-0451-4000-b000-000000000000: Unknown
	[Characteristic] 0000ffd6-0000-1000-8000-00805f9b34fb: (read) | Name: Log Retrieval, Value: b'\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f\x1f' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 12) | Value: b'\x00\x00' 
		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb: (Handle: 13) | Value: b'Log Retrieval'

dlech, 2020-05-13:
Can you please fill in all of the version information?

gyamanishi, 2020-05-13:
Here is another clue.  I can read a 200 byte descriptor no problem.  When I try to read a 200 byte value I get: 
	[Characteristic] 0000fff5-0000-1000-8000-00805f9b34fb: (read) | Name: Characteristic 5, Value: b'Could not read characteristic value for 0000fff5-0000-1000-8000-00805f9b34fb: 2'
		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb: (Handle: 30) | Value: b'Characteristic 5'

hbldh, 2020-05-13:
This is interesting. The `:2` at the end signifies that the [`GattCommunicationStatus`](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcommunicationstatus?view=winrt-19041) returned is a `ProtocolError` meaning _There was a GATT communication protocol error._

Thus, the Windows backend answered that this characteristic cannot be read and the error did not originate in bleak. Have you tried to read it in a Linux or macOS environment?

gyamanishi, 2020-05-13:
no.  Not really setup for linux or mac at the moment.

hbldh, 2020-05-15:
Try using the descriptor instead, see it it changes anything.

It might be a TI issue. I cannot say really. Can it really handle the large write that you did? Just because it does not respond with an error, does it mean that everything went ok? If you write 244 bytes, is it guaranteed that you can read them back from the same characteristic? I cannot help you any further here, I am afraid.

hbldh, 2020-07-20:
I think this is closed by changes described in #217 and the bumped `pythonnet` version in #232.

closed 2020-07-20 by hbldh