hbldh/bleak #190: set_disconnected_callback does not work on CoreBluetooth

niemyjski, 2020-05-14:
* bleak version: latest
* Python version: 3.8
* Operating System: macOS Latest
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

Unable to set disconnected callback on macOS.

### What I Did

```
def onDisconnect(client):
    print("Client with address {} got disconnected!".format(client.address))

async with BleakClient(device.address, loop=loop) as client:
        try:
            client.set_disconnected_callback(onDisconnect)
```

dlech, 2020-05-14:
Duplicate of #184?

hbldh, 2020-05-15:
You are should be unable to do this right now and bleak should throw a `NotImplementedError` with the text `"This is not implemented in the Core Bluetooth backend yet"` which should tell you everything you need to know.

Apart from that, a PR implementing this is in the works: #186

closed 2020-05-15 by hbldh