hbldh/bleak #191: Connecting via BleakClient with address does an additional discovery.

niemyjski, 2020-05-14:
* bleak version: latest
* Python version: 3.8
* Operating System: macOS 
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I noticed it was taking about 5 seconds to discover devices and then connect to a discovered device (another 5 seconds)... After looking at the code it appears that the client implementation does a second discovery which is painfully slow. Is there any way we can just pass in a populated device or just connect to the uuid directly instead of rediscovering?

https://github.com/hbldh/bleak/blob/08f4fc4cfb9b8c334b31d1e0ee76b6088684ec44/bleak/backends/corebluetooth/client.py#L63

### What I Did

```
async def discoverDevice() -> BLEDevice:
    devices = await discover()
    for d in devices:
        name = d.details.name()
        if name is not None and "TEST" in name:
            return d

    return None

async def run(loop):
    device = await discoverDevice()
    if device is None:
        return int(-1)

    # this does another discovery..
    async with BleakClient(device.address, loop=loop) as client:

loop = asyncio.get_event_loop()
loop.run_until_complete(run(loop))
```

hbldh, 2020-05-15:
Yes, if you look at the [BleakClientCoreBluetooth](https://github.com/hbldh/bleak/blob/master/bleak/backends/corebluetooth/client.py#L36-L37), there is a `timeout` keyword to use for this very purpose.

A `discover` has to have been called before maing any connect, that is why it is in `connect` to ensure that the native OS backend actually has the BLE device in its' caches.

dhalbert, 2020-06-07:
> A `discover` has to have been called before maing any connect, that is why it is in `connect` to ensure that the native OS backend actually has the BLE device in its' caches.

I encountered the same thing just today. As @niemyjski mentioned, if `BleakClient()` alternatively took a `device` instead of an address, it could assume the device had been recently discovered. The constructor could take either, and do a `discover()` only if an address were passed.

niemyjski, 2020-06-08:
That would be really nice :)

dlech, 2020-09-10:
I agree that this is something that should be fixed.

hbldh, 2020-09-10:
@dhalbert That is a good suggestion. I had planned on accepting `BLEDevice`s as well as strings in the `__init__` and skip the second discovery in the 0.8.0 release. Somehow I missed implementing it, thinking that I already had done it. 😞

hbldh, 2020-09-25:
This is fixed in release 0.8.0. Closing.

PR #285 (2020-09-14): Connect by BLEDevice as well as string Bluetooth adress
* Added the possibility to connect using ``BLEDevice`` instead of a string adress. This allows for skipping the discovery call when connecting.
* Renamed ``BleakClient`` argument ``address`` to ``address_or_device``. 

Fixes #191 
Improves #284
closed 2020-09-25 by hbldh