hbldh/bleak #203: Can't connect to device unless it is in pairing mode

jcgfsantos, 2020-05-20:
* bleak version: 0.6.4
* Python version: 3.6.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I can't connect to a device unless it is in pairing mode although it shows up when scanning for nearby devices.
Once in pairing mode I can subscribe a characteristic and receive it's value. Once the pairing mode in the device stops I stop receiving the value and get an error:

Disconnecting from BLE device...
Traceback (most recent call last):
  File "getTH.py", line 173, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "\AppData\Local\Programs\Python\Python36-32\lib\asyncio\base_events.py", line 484, in run_until_complete
    return future.result()
  File "getTH.py", line 87, in run
    await client.start_notify(CHARACTERISTIC_UUID, notification_handler)
  File "\AppData\Local\Programs\Python\Python36-32\lib\site-packages\bleak\backends\dotnet\client.py", line 466, in start_notify
    "Could not start notify on {0}: {1}".format(characteristic.uuid, status)
bleak.exc.BleakError: Could not start notify on 226caa55-6476-4566-7562-66734470666d: 3

hbldh, 2020-05-25:
The `: 3` means that [access is denied to that characteristic](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcommunicationstatus?view=winrt-18362).

This relates to all other issues with devices that have security restrictions in their devices. Bleak cannot handle this yet, and I have no roadmap as to when it might have.

obzerving, 2020-09-06:
bleak version: 0.7.1
Python version: 3.7.7
Operating System: Windows 10
Backend: .Net

I don't know if this will muddy the waters, but I also can't connect to a device (an ESP-32 in particular) unless I pair it first. Otherwise, if I run a program like get_services.py, I'll get an error message something like device with address XXX not found. However, if I then remove it from the list of bluetooth devices known by Windows 10, get_services.py will successfully run even though the device isn't in the list or in the Device Manager. This situation persists over system restarts. I checked through the Windows 10 registry, but didn't see any signs that it was remembering the device's name or address.

hbldh, 2020-09-10:
@obzerving In the `develop` branch, there is pairing functionality prepared, but it requires a connection to be made first. Try installing from the `develop` branch or wait until release of 0.8.0 (sometime next week I think) and try again then.

obzerving, 2020-09-10:
Thank you for your reply and for all the effort you've put into this software.

dlech, 2022-07-25:
closing due to inactivity

PR #269 (2020-09-02): First drafts of "Just Works" pairing for .NET and BlueZ backends
Might fix #64
Might fix #77
Might fix #203 
Fixes #218
closed 2022-07-25 by dlech