hbldh/bleak #204: No Connection Timeout

GitChris3004, 2020-05-25:
* bleak version: 0.6.2
* Python version: 3.7.4
* Operating System: Windows 10

### Description
There is no timeout for connecting, only for discovering.

### What I Did
When I try to connect to BLE device which doesn't respond properly with that code:
` 
async with bleak.BleakClient(device.address, loop=loop_asynchio, timeout=5.0) as self.connected_ble_client:
`
 bleak get stuck in the BleakClientDotNet class in the _ConnectionStatusChanged_Handler function in the line:         
`services = await self.get_services()`
It would be nice to have a timeout setting for this part of code. Otherwise there is no other possibility than waiting until the function fails. Which takes pretty long.

hbldh, 2020-05-25:
Interesting. I have never experienced that anything hangs there. That means that a connection has been established, but that services cannot be fetched. The Windows Bluetooth stack should (and have in my experience) sent a timeout and error here.

hbldh, 2020-09-25:
Not sure if this is still an issue. Will close for now.

closed 2020-09-25 by hbldh