hbldh/bleak #206: MacOS: scanning in new thread raises BleakError("Bluetooth device is turned off")

dhalbert, 2020-05-25:
* bleak version: 0.6.4
* Python version: 3.8.2
* Operating System: MacOS 10.15.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

Trying to do scanning in a new thread raises `BleakError("Bluetooth device is turned off")`. Run the program below with and without `--thread`. In the latter case, the error is raised, but the program works fine with everything in the same thread.

Using `BleakScanner` rather than `discover` has the same problem.

This program runs fine on Linux with BlueZ.

This sounds sort of like #93 and #125, but those were BlueZ problems. This stackoverflow Q&A may also be of interest, though I don't know corebluetooth well enough to confirm: https://stackoverflow.com/questions/48958267/how-can-i-use-corebluetooth-for-python-without-giving-up-the-main-thread

The reason I'm using a second thread is that I'm re-implementing an existing non-`async` BLE API using bleak. I was using a fresh thread for the bleak `async` side of things, and a `janus` queue to pass back scanning results.

```python3
import asyncio
from bleak import discover
from threading import Thread
import argparse

parser = argparse.ArgumentParser()
parser.add_argument('--thread', action='store_true')
args = parser.parse_args()

async def scan():
    devices = await discover()
    for d in devices:
        print(d)

def run():
    asyncio.run(scan(), debug=True)

if args.thread:
    print("in separate thread")
    t = Thread(target=run)
    t.start()
    t.join()
else:
    print("in main thread")
    loop = asyncio.get_event_loop()
    loop.run_until_complete(scan())
```

cspensky, 2020-05-25:
I have seen this same behavior on my Mac, trying to solve a similar
problem.

hbldh, 2020-06-02:
Relates to #105. On macOS , one should try to avoid using several threads/event loops in bleak, since the use of a global `CentralManagerDelegate` will produce problems. Run the bleak programs in different processes if possible and it _should_ work.

This needs to be put into the documentation about the macOS backend.

LTKI, 2020-06-02:
I have the same problem. Even when using a different process for bleak, I get the same error. I tried both "spawn" and "fork". Do I understand it correctly that it is therefore not possible to execute bluetooth actions in the background of your program, e.g. waiting for a BLE notify while showing a loading bar to the user/running a gui? Maybe you could point us in the right direction for a fix?

dhalbert, 2020-06-08:
More data: even just a new event loop (as you implied/mentioned) causes the problem, without using a new thread. Using `asyncio.run()`, which creates a new event loop, instead of `loop.run_until_complete()` on the existing event loop, causes the same problem.

```python
import asyncio
from bleak import discover
from threading import Thread
import argparse

parser = argparse.ArgumentParser()
parser.add_argument('--thread', action='store_true')
parser.add_argument('--run_until_complete', action='store_true')
parser.add_argument('--run', action='store_true')
args = parser.parse_args()

async def scan():
    devices = await discover()
    for d in devices:
        print(d)

def run():
    asyncio.run(scan(), debug=True)

if args.thread:
    print("in separate thread")
    t = Thread(target=run)
    t.start()
    t.join()

if args.run_until_complete:
    print("run_until_complete in main thread")
    loop = asyncio.get_event_loop()
    loop.run_until_complete(scan())

if args.run:
    print("run in main thread")
    asyncio.run(scan())
```

dhalbert, 2020-06-08:
Looking at `backends/corebluetooth/__init__.py`, I see  that the `CBAPP` singleton of class `Application` is created on import, and that it grabs the current event loop. https://github.com/hbldh/bleak/blob/9c87ac06a7f04e9d36c553cfcf51f4b955d2c03a/bleak/backends/corebluetooth/__init__.py#L26-L27,

I think a simple fix would be to lazily create the singleton when it's actually referenced for the first time, and have it use either the current event loop at that time, or perhaps optionally the loop passed in to `BleakClient`, etc. That would accommodate the simple cases in the comments above where a single new event loop is created for all bleak operations.

I'll see if I can work up a PR for this. I see several un-merged corebluetooth PR's, but they don't appear to change how the singleton is created.

superfashi, 2020-06-12:
The following code doesn't even seem to work, which I found really confusing:
```python
import asyncio

async def run():
    from bleak import discover
    print(await discover())

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

hbldh, 2020-06-24:
This issue will be worked on 2020-06-30, so any additional information that can be added here before that is very welcome.

dhalbert, 2020-06-24:
> This issue will be worked on 2020-06-30, so any additional information that can be added here before that is very welcome.

I'm not sure how useful this would be, but I'll explain the context for my original report. We are using bleak to (partially) reimplement [`_bleio` ](https://circuitpython.readthedocs.io/en/latest/autoapi/_bleio/index.html), an existing API for CircuitPython. CircuiPython is a friendly fork of MicroPython, a mostly bare-metal implementation of Python for microcontrollers. `_bleio` was originally designed for the nRF52840 native SoftDevice BLE stack. But we also want to run code written for CircuitPython on CPython on host computers, so we have created various wrapper libraries that go by the name "Blinka", in this case [Adafruit_Blinka_bleio](https://github.com/adafruit/Adafruit_Blinka_bleio). I have Blinka _bleio running on x64 and Raspbian Linux,  and Windows 10. Getting it to run on MacOS will complete `_bleio` support on the three major host platforms, in addition to the original CircuitPython implementation. I am also starting on third implementation of `_bleio` for direct use of HCI-capable co-processors.

The more identically things work on the these four platforms, the better. We want a single program to be runnable across all four. Obviously some things don't work, such as acting as a peripheral. We also have issues with scanning: the bluez kernel support does de-duplication of advertisements, which is undesirable when advertising is used for changing data transmissions, so we have to use privileged `hcitool` and `hcidump` right now on Linux for scanning, and fall back to bleak only if they are not available.

`_bleio` is not meant for end-user use: instead we have a nicer wrapper library, [`adafruit_ble`](https://github.com/adafruit/Adafruit_CircuitPython_BLE). The `examples` directory there contains simple examples, and then we also  have [a large number of individual libraries that support various devices](https://github.com/adafruit?q=adafruit_circuitpython_ble&type=&language=).

dlech, 2020-06-24:
I had the same error as the title of this issue without using threads. The changes I made in https://github.com/pybricks/bleak/commit/26ab00f4d619f1db2618462a70b40e85bbcb3dd1 and https://github.com/pybricks/bleak/commit/0b37282af230c0b1161daaaa399a5bf328f28681 fixed the issue for me. But ideally it would be nice to get rid of the global `CBAPP` object. It causes other os-specific quirks like #111.

(@dhalbert in case you haven't seen, it looks like BlueZ is getting better advertisement support: https://github.com/bluez/bluez/blob/master/doc/advertisement-monitor-api.txt - might be good to make sure it does what is needed while it is still experimental)

cspensky, 2020-06-24:
> changes I made in pybricks@26ab00f
> and pybricks@0b37282

I also ran into this issue and fought with it quite a bit.  I was able to
get the error to go away after some hackery, but was never actually able to
achieve full functionality of bleak in a thread.  Happy to help any way I
can, this is also a high priority item for me.  I was planning to knock it
out in July if you don't come up with a fix by then.

dlech, 2020-06-26:
FYI, I've added https://github.com/pybricks/bleak/commit/1b04383537825b103a325f7a3d395ab9875a3e34 and https://github.com/pybricks/bleak/commit/f11a2e1c54a3f18ab4c293b4e17c4e620f410b82 to our fork which gets the example program in https://github.com/hbldh/bleak/issues/206#issue-624413861 working in the thread.

Still a bit of work left to do to reconcile it with the other proposes changes to the corebluetooth backend, but it probably wouldn't hurt to try it out if anyone is interested.

hbldh, 2020-06-30:
I have tested this issue both with threads and with event loops after merging #227 to `develop` and it seems to be a solid solution. This will be released later today.

closed 2020-06-30 by hbldh

hbldh, 2020-06-30:
@dhalbert I have looked at `_bleio`; very nice! It would be great if bleak could cater nicely to that kind of environment as well! Hitherto, it has been focused on desktop only, with some at least support for Raspberry Pi.

The scanning issue that you are mentioning, could you please open a separate issue for that, because that is something I did not know about. I have been getting `PropertiesChanged` notifications in from BlueZ during scanning, but that might not yield new data, only updated RSSI values?

epoz, 2021-05-05:
FWIW, doing a "pip install bleak" on OS X  (10.14.6) and trying the first example from the Usage section of the README still fails with an error:

```
Traceback (most recent call last):
  File "x", line 10, in <module>
    loop.run_until_complete(run())
  File "/usr/local/opt/python@3.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "x", line 5, in run
    devices = await BleakScanner.discover()
  File "/Users/etienne/envs/p3/lib/python3.7/site-packages/bleak/backends/scanner.py", line 83, in discover
    async with cls(**kwargs) as scanner:
  File "/Users/etienne/envs/p3/lib/python3.7/site-packages/bleak/backends/corebluetooth/scanner.py", line 37, in __init__
    self._manager = CentralManagerDelegate.alloc().init()
  File "/Users/etienne/envs/p3/lib/python3.7/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 93, in init
    raise BleakError("Bluetooth device is turned off")
```

While Bluetooth is definitely on. Googling this error message brought me to this issue, so not sure if this could still be a thing, even though this issue has been closed.

commit ad2f8eb4 (2020-06-30): Changelog entry about fixes in #227.

Fixes #105
Fixes #111
Fixes #211
Fixes #206
PR #232 (2020-06-30): Release/v0.7.0
# Version 0.7.0 (2020-06-30)

## Added

* Better feedback of communication errors to user in .NET backend and implementing error details proposed in #174.
* Two devices example file to use for e.g. debugging.
* Detection/discovery callbacks in Core Bluetooth backend ``Scanner`` implemented.
* Characteristic handle printout in ``service_explorer.py``.
* Added scanning filters to .NET backend's ``discover`` method.

## Changed

* Replace ``NSRunLoop`` with dispatch queue in Core Bluetooth backend. This causes callbacks to be dispatched on a background thread instead of on the main [... cut]