hbldh/bleak #211: Unable to run scan example: 'CentralManagerDelegate' object has no attribute 'peripherals'

f0rodo, 2020-06-03:
* bleak version: 0.6.1
* Python version: 3.7
* Operating System: MacOSX 10.15.4 (19E287) Catalina

### Description

Using Bleak scanning example I ran into an attribute issue.

### What I Did

Scan.py:
```
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

```
python scan.py
Traceback (most recent call last):
  File "scan.py", line 19, in <module>
    loop.run_until_complete(run())
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "scan.py", line 13, in run
    devices = await scanner.get_discovered_devices()
  File "/tmp/.venv/lib/python3.7/site-packages/bleak/backends/corebluetooth/scanner.py", line 64, in get_discovered_devices
    peripherals = cbapp.central_manager_delegate.peripheral_list
AttributeError: 'CentralManagerDelegate' object has no attribute 'peripheral_list'
```

bsiever, 2020-06-10:
Just to double check, this is fixed by PR #209, right?

f0rodo, 2020-06-10:
yup! @bsiever

hbldh, 2020-06-24:
Good that this is solved already!

The `peripheral_list` attribute is something I think exists on my old Mac Mini running El Capitan, because it works there but has apparently been changed since. #209 will be merged in version 0.7.0.

commit ad2f8eb4 (2020-06-30): Changelog entry about fixes in #227.

Fixes #105
Fixes #111
Fixes #211
Fixes #206
closed 2020-06-30 by hbldh