hbldh/bleak #214: Using start_notify() and transmition stopped

SunBeam4779, 2020-06-08:
* bleak version: 0.6.4
* Python version: 3.6.5
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I used a BLE board which is a TI's launchPad to transmit data to my computer. 
I'm going to use bleak to make a little app which can accept and save the BLE notification.
I start with the code:
async with BleakClient(address_, loop=loop_) as client:
    await client.start_notify(indication_characteristic_uuid, data_handler)
    await stop_event.wait()
    await client.stop_notify(indication_characteristic_uuid)
I run this code and the connection is successful, and the notification could be accepted. I expect that the notification could be accepted continuously until I stop my BLE device. But after getting several bytes, both the code and my BLE device stopped.
In the app BLE utility which is on my android phone, this never happened. So I think the issue might not be my device's.

### What I Did

async def run(address_, loop_, data_storage, debug=False):
    log = logging.getLogger(__name__)
    stop_event = asyncio.Event()

    def data_handler(sender, data_):
        log.info("{0}: {1}".format(sender, data_))
        data_storage.append((datetime.datetime.utcnow().timestamp(), data_))

    async with BleakClient(address_, loop=loop_) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))

        log.info("Starting notifications/indications...")
       
        await client.start_notify(indication_characteristic_uuid, data_handler)

        await stop_event.wait()

        await client.stop_notify(indication_characteristic_uuid)
        log.info("Stopping notifications/indications...")

    log.info("Disconnected")

Here is the output:
D:\python3\python.exe "D:/python3/workspace/BLE Connector/BLE Chat.py"
Executing <Handle <TaskWakeupMethWrapper object at 0x000001DBF27AB528>(<Future finis...events.py:284>) created at D:\python3\lib\asyncio\futures.py:343> took 0.141 seconds
Connected: True
Starting notifications/indications...
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x00\xc0\x7f\xff\x03j\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03j\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03m\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03o\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03n\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03m\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03m\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03n\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03n\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03q\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03p\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff')
[... 12 lines cut ...]
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03m\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03o\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03o\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03n\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03p')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff\x03o\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03n\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03m\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03k\xc0\xc0\x7f\xff\x03k\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03l\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff\x03n')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03n\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x7f\xff\x03o\xc0\xc0\x7f\xff\x03o\xc0\xc0\x7f\xff\x03p\xc0\xc0\x7f\xff')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x03p\xc0\xc0\x7f\xff\x03n\xc0\xc0\x7f\xff\x03m\xc0\xc0\x7f\xff\x03m')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03l\xc0\xc0\x7f\xff\x03k\xc0\xc0')
f000fff1-0451-4000-b000-000000000000: bytearray(b'\x00D\x00 \xb1W\x01\x00\x01\xc9\x01\x10\x01\xc9\x01\x10\x01\xc9\x01\x10')

and it just stopped.

Could you help me find out where the issue is?

SunBeam4779, 2020-06-08:
I'm sorry about saying that the issue might not be my BLE device.
Actually, I figured out the problem just now.
It's all my faul that the BLE device code makes the length of bytes too small causing the blocking of data transmit.
bleak is a wonderful project.

Ladvien, 2020-06-11:
Hey @SunBeam4779,

Maybe this issue could be closed out for clarity's sake?

closed 2020-06-11 by SunBeam4779

tomuGo, 2021-02-21:
> 我很抱歉地说问题可能不是我的BLE设备。
> 实际上，我刚刚发现了问题。
> BLE设备代码使字节长度太小导致数据传输阻塞，这是我的全部错。
> 惨淡是一个了不起的项目。

hi. same ,could u tell me how to solved it.think u

tomuGo, 2021-02-21:
> I'm sorry about saying that the issue might not be my BLE device.
> Actually, I figured out the problem just now.
> It's all my faul that the BLE device code makes the length of bytes too small causing the blocking of data transmit.
> bleak is a wonderful project.

hi. same ,could u tell me how to solved it.think u