hbldh/bleak #216: Error "Attempt to disconnect device failed: Connection was closed cleanly." causes async client context to never exit

magnusnordlander, 2020-06-14:
* bleak version: 0.6.4
* Python version: 3.7.3
* Operating System: Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux / Raspbian GNU/Linux 10 (buster)
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.50

### Description

I've written some sample code to demonstrate:

```python
import sys
import asyncio
import logging
from bleak import BleakClient

logging.root.setLevel(logging.DEBUG)
handler = logging.StreamHandler(sys.stdout)
handler.setLevel(logging.DEBUG)
FORMAT = "%(asctime)-15s %(name)-8s %(levelname)s: %(message)s"
handler.setFormatter(logging.Formatter(fmt=FORMAT))
logging.root.addHandler(handler)

async def run(loop):
    print("Running")
    async with BleakClient("9B0F0A71-568C-4DB5-9002-F1D09B240D0A" if sys.platform == "darwin" else "00:1c:97:1a:a0:2f", loop=loop) as client:
        print("Created")
        await client.is_connected()
        print("Connected")
        await asyncio.sleep(3)
        await client.disconnect()
        print("Disconnected")
    print("Done")

loop = asyncio.get_event_loop()
loop.run_until_complete(run(loop))
loop.run_forever()
```

So, I'm connecting to an Acaia Lunar scale, or rather disconnecting from one, and on Linux, disconnecting gives me a weird error (`ERROR:bleak.backends.bluezdbus.client:Attempt to disconnect device failed: Connection was closed cleanly.`) and then the BleakClient context never exits. On OS X it works fine.

### What I Did

Output on OS X:
```
/Users/magnusnordlander/Developer/silvia-pi/venv/bin/python /Users/magnusnordlander/Developer/silvia-pi/BleakTest.py
2020-06-14 23:33:20,364 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
Running
2020-06-14 23:33:20,381 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 3D963F76-BE4B-4600-ADA1-C289B2E476A6: Unknown @ RSSI: -80 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,382 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 3D963F76-BE4B-4600-ADA1-C289B2E476A6: Unknown @ RSSI: -80 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,392 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 2865E8B9-A672-420A-8692-E720493E11F9: Unknown @ RSSI: -72 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,394 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 2865E8B9-A672-420A-8692-E720493E11F9: Unknown @ RSSI: -72 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
2020-06-14 23:33:20,400 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 3FFCE2ED-2AC5-4FD8-84FF-0F4E8802CFF2: Unknown @ RSSI: -86 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,408 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9B8CC7D2-EF3B-46E9-95A2-0360F1DD7073: HK Onyx Studio 4 @ RSSI: -87 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,411 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 31CBF837-B976-4FFC-A84B-DD2A86016A8C: Unknown @ RSSI: -68 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,413 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device E33DEB5D-6ECC-4052-A262-49627F7B7356: Unknown @ RSSI: -82 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,414 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device E33DEB5D-6ECC-4052-A262-49627F7B7356: Unknown @ RSSI: -82 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,418 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 50785798-19F9-4EF0-B0BE-59AA822B5351: Unknown @ RSSI: -49 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,423 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 7346DF6F-6953-4D59-94C7-D0C0659D220D: Unknown @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,424 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 7346DF6F-6953-4D59-94C7-D0C0659D220D: Unknown @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,427 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 522F6BC9-7E4A-4F7D-9AA2-7BE1923B8049: Unknown @ RSSI: -94 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,429 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 522F6BC9-7E4A-4F7D-9AA2-7BE1923B8049: Unknown @ RSSI: -94 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
2020-06-14 23:33:20,491 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device FA2EE6B3-9775-442A-92BA-BC0C0B49BADB: Unknown @ RSSI: -52 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,492 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device FA2EE6B3-9775-442A-92BA-BC0C0B49BADB: Unknown @ RSSI: -52 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,508 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9B8CC7D2-EF3B-46E9-95A2-0360F1DD7073: HK Onyx Studio 4 @ RSSI: -85 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,768 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 61C334B4-CB32-40FF-881A-EE12EE81ADE5: ScootyPuffJr @ RSSI: -89 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,771 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 61C334B4-CB32-40FF-881A-EE12EE81ADE5: ScootyPuffJr @ RSSI: -89 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,773 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 47F3B9C7-C9EC-4BD0-B7F7-4B4A8DF1449D: Unknown @ RSSI: -55 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,774 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 47F3B9C7-C9EC-4BD0-B7F7-4B4A8DF1449D: Unknown @ RSSI: -55 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-06-14 23:33:20,777 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9B0F0A71-568C-4DB5-9002-F1D09B240D0A: ACAIAL1A02F @ RSSI: -41 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
2020-06-14 23:33:20,778 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 9B0F0A71-568C-4DB5-9002-F1D09B240D0A: ACAIAL1A02F @ RSSI: -41 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
[... 147 lines cut ...]
	00001820-0000-1000-8000-00805f9b34fb
	Internet Protocol Support
2020-06-14 22:34:35,369 bleak.backends.bluezdbus.client DEBUG: 
Characteristic
	/org/bluez/hci0/dev_00_1C_97_1A_A0_2F/service000b/char000c
	00002a80-0000-1000-8000-00805f9b34fb
	Age
2020-06-14 22:34:35,374 bleak.backends.bluezdbus.client DEBUG: 
Descriptor
[... cut ...]
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
Created
Connected
2020-06-14 22:34:38,486 bleak.backends.bluezdbus.client DEBUG: Disconnecting from BLE device...
2020-06-14 22:34:38,489 bleak.backends.bluezdbus.client DEBUG: Removing rule PropChanged, ID: 1
Disconnected
2020-06-14 22:34:38,842 bleak.backends.bluezdbus.client DEBUG: Disconnecting from BLE device...
2020-06-14 22:34:38,864 bleak.backends.bluezdbus.client ERROR: Attempt to disconnect device failed: Connection was closed cleanly.
```

magnusnordlander, 2020-06-15:
Okay, so I didn't realize that `__aexit__` actually calls `disconnect`. So basically, this is just a matter of not being able to call `disconnect` twice when using bluez, whereas that's possible using CoreBluetooth.

Ladvien, 2020-06-18:
Hey @BillMicro,

I woke up this morning and realized you'd closed this issue and posted on the #219 out of confusion.  Doh.  

Anyway, after the above code changed I provided I realize the usage of a `BleakClient` context manager is understood to be a _connected_ client.  That stated, I don't think managing the connection inside the context is intended.  

If manual connection management is needed, then getting a reference to the instantiated `BleakClient` would be the way to go.

Anyway, I write this out for posterity.

Ladvien, 2020-06-19:
Forgot to ask, is it ok to close out this issue now?

BillMicro, 2020-06-19:
Hey @Ladvien, I posted to #221, where someone raised a very similar question - I think @magnusnordlander thinks this issue is now resolved, so maybe you can close it. I'm also new to this project and I'm not an admin. 

I guess I could try using manual connection management in order to deal with the issue I posted about in #219 - I might give that a try today.

magnusnordlander, 2020-06-20:
I haven't actually installed it, but it looks like 8794eac would solve this too, yeah. I'll close the issue once I've had the opportunity to verify.

hbldh, 2020-06-24:
This inconsistency will be addressed shortly and a fix will be included in version 0.7.0. Thank you for reporting it!

commit c267f9a4 (2020-06-30): Improved handling of txdbus connection object at disconnect in BlueZ backend.

Fixes #216
Fixes #219
Fixes #221
closed 2020-06-30 by hbldh