hbldh/bleak #218: How to register custom Agent (device pairing with BlueZ)

bojanpotocnik, 2020-06-16:
* bleak version: 0.6.5a3 (latest develop)
* Python version: 3.8.2
* Operating System: Ubuntu 20.04 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

I would like to implement pairing, but it seems that my custom Agent does not work - callback methods (RequestPasskey, DisplayPasskey, RequestPinCode,... any) are never invoked.

1. I created [custom Agent class](https://github.com/hbldh/bleak/compare/develop...bojanpotocnik:develop-bluez_pairing?expand=1#diff-01fef9e4e1599d35320daccd4c5a24a1R36),
2. [registered it](https://github.com/hbldh/bleak/compare/develop...bojanpotocnik:develop-bluez_pairing?expand=1#diff-01fef9e4e1599d35320daccd4c5a24a1R422-R425)
3. and then invoked [Pair instead of Connect](https://github.com/hbldh/bleak/compare/develop...bojanpotocnik:develop-bluez_pairing?expand=1#diff-01fef9e4e1599d35320daccd4c5a24a1R449).
4. I tried all combinations of method signatures and at the end even put [two different signatures together](https://github.com/hbldh/bleak/compare/develop...bojanpotocnik:develop-bluez_pairing?expand=1#diff-01fef9e4e1599d35320daccd4c5a24a1R130-R150), but still the methods are never invoked.

Note that using "raw DBus interface" - [bluez_dbus_utils.py](https://gist.github.com/bojanpotocnik/2049e4914ea581d6814019992850743d#file-bluez_dbus_utils-py) code with snippet:
```python
import dbus
from bluez_dbus_utils import Agent, Adapter, bluez

dbus.mainloop.glib.DBusGMainLoop(set_as_default=True)

adapter = Adapter()
agent = Agent()

device = adapter.find(re.compile(r"AA:BB:CC:[\d:]+"))

print("Pairing...")
agent.passkey = 32699
response = device.pair()
print("Paired!", type(response), response)

print("Running...")
try:
    bluez.mainloop_run(timeout=5)
except KeyboardInterrupt:
    pass
print("Finnish")

device.Disconnect()
bluez.mainloop_run(until_idle=True)
```
works properly (device is paired and stays connected), so I know what and how it shall be done, but I would like to implement this in `bleak`. Seems that I have problems with `txdbux` style of implementing custom DBus objects?

hbldh, 2020-06-24:
This is very interesting work, but far outside the scope of what I can help you with right now. I am not sure a custom Agent is the proper way to go about this; does not the regular pairing methods work? Besides, the `dbus` library is not something I will use with bleak, since it is not `asyncio` compatible from what I know. Use `txdbus` instead.

The reason i question this is that the API needs to be able to work on BlueZ, macOS and Windows. This seems to make the BlueZ implementation too swollen and complex for my liking.

bojanpotocnik, 2020-06-25:
I understand - I would like to do it using txdbus, but I am not able to. That's why I have done it with "raw" dbus library first to see that it works and that I should probably be able to do it using txdbus. Otherwise, Agent is meant for such purposes and each application can register it's own agent - [Bluez docs: Agent](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/agent-api.txt).

> does not the regular pairing methods work?

Why do you mean by this? I would be much pleased if there is simpler way of doing this.

hbldh, 2020-06-30:
The `dbus` library uses the `glib` event loop which does not play nice with `asyncio` from what know. I will not try to integrate it with bleak, so if you want to get something into Bleak, it would be more wothwhile for you to try something else.

By "regular" pairing I mean something like this:
- Windows: https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/pair-devices
- BlueZ: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt#n88
- Core Bluetooth: No idea right now...

hbldh, 2020-09-25:
Pairing in BlueZ and .NET is implemented in version 0.8.0. Closing this.

PR #269 (2020-09-02): First drafts of "Just Works" pairing for .NET and BlueZ backends
Might fix #64
Might fix #77
Might fix #203 
Fixes #218
closed 2020-09-25 by hbldh