hbldh/bleak #220: Connect to an already-paired device on Mac

paoloaveri, 2020-06-17:
* bleak version: 0.6.4
* Python version: Python 3.7.6
* Operating System: MacOs 10.14.5 Mojave

### Description

I noticed that I could not connect to a BLE device that was already paired to my Mac and showing up in System Preferences -> Bluetooth. The discover() method doesn't find it. 

### What I Did
In Core Bluetooth there is a retrieveConnectedPeripheralsWithServices method that gives a list of connected devices that have a certain service. So I added a get_connected_by_services() method in discovery.py. That allows me to search for my device in the list of connected devices first, and then do a discover() if not found. I changed the connect() method to also look at the list of connected devices first and do a discover() if not found. Please see https://github.com/paoloaveri/bleak/commit/a4ad3a9f01beac03d98623bb08c661c708431503
Sadly I didn't find any method to get the list of all connected devices, so I have to use retrieveConnectedPeripheralsWithServices and provide at least one service that I know my device have.

This is a working script to update a smart watch over BLE: https://github.com/paoloaveri/openp8-research/blob/master/python-update/p8_update.py. It is working on Mac only, with the above modified version of bleak (https://github.com/paoloaveri/bleak/commit/a4ad3a9f01beac03d98623bb08c661c708431503).

Here is an extract of how I use the new get_connected_by_services():
```
# Check if the device is already connected
services = ["FEEA"]
devices = await get_connected_by_services(service_uuids=services)
for d in devices:
    if(d.name == "P8 a"):
        print("P8 watch is already connected, great!!")
        gb_p8_address = d.address
# If not already connect, scan for it
if not gb_p8_address:
    devices = await discover()
    for d in devices:
        if(d.name == "P8 a"):
            print("P8 watch was found!!")
            gb_p8_address = d.address
```

hbldh, 2020-06-24:
Hm. Unfortunately, Bleak does not handle any kind of pairing right now, and I cannot help you with what happens if your device is already paired outside of bleak. It is too much unknown right now.

I intend to address pairing eventually, but right now, it is a non-supported feature.

paoloaveri, 2020-06-27:
I have to say it's pretty useful to be able to connect to a paired device when you do testing and run a bleak script dozens of times. Without my modifications, I would have to manually remove the BLE device in System Preferences between each run.

This is also a behavior that doesn't seem to be consistent over different platforms. I tested the same thing on a Raspberry Pi 3, and it is possible to connect to a paired device over there, the discover() method finds it. So maybe it's an issue with the CoreBluetooth API.

But I understand, thank you for your answer.

hbldh, 2020-09-25:
The connection situation in the Core Bluetooth backend is greatly improved in version 0.8.0. This might be resolved there. Will close for now, reopen if it is still present.

closed 2020-09-25 by hbldh