hbldh/bleak #221: disconnect() hangs for ever on Linux (Raspberry Pi 3)

paoloaveri, 2020-06-17:
* bleak version: 0.6.4
* Python version: Python 3.7.3
* Operating System: Raspbian GNU/Linux 10 (buster) (Linux pi3 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux)
* BlueZ version: 5.50
* On Raspberry Pi 3 Model B+

### Description
Calling the disconnect method makes my python script hang for ever. After investigating with ipdb, the problem is in `disconnect(self)` line 238 in `bleak/backends/bluezdbus/client.py`: `await self.is_connected()` is hanging for ever:
```
# See if it has been disconnected.
is_disconnected = not await self.is_connected()
```
### What I Did
This is a python script I'm using to reproduce the issue
```
import asyncio
from bleak import *

async def run():
    print("Searching for P8 watch...")
    devices = await discover()
    for d in devices:
        if(d.name == "P8a"):
            print("P8 watch was found!!")
            gb_p8_address = d.address
    if(gb_p8_address == None):
        print("P8 wasn't found, sorry!")
        exit()

    async with BleakClient(gb_p8_address, loop=asyncio.get_event_loop()) as client:
        is_connected = await client.is_connected()
        print(f"Device connected?: {is_connected}")

        gatt = await client.read_gatt_char("00002a24-0000-1000-8000-00805f9b34fb")
        print(gatt)

        print("disconnecting...")
        await client.disconnect()

        while await client.is_connected():
            print("waiting for device to disconnect...")
            await asyncio.sleep(1)

    print("Trying to reconnect...")
    async with BleakClient(gb_p8_address, loop=asyncio.get_event_loop()) as client:
        # Do other stuff
        exit()

if __name__ == "__main__":
    logging.basicConfig(stream=sys.stdout, level=logging.DEBUG)
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```
And this is the output of the script:
```
DEBUG:asyncio:Using selector: EpollSelector
Searching for P8 watch...
INFO:bleak.backends.bluezdbus.discovery:P8a, FD:67:23:AA:07:AF (-48 dBm), Object Path: /org/bluez/hci0/dev_FD_67_23_AA_07_AF
[...lots of devices...]
P8 watch was found!!
INFO:bleak.backends.bluezdbus.discovery:P8a, FD:67:23:AA:07:AF (-48 dBm), Object Path: /org/bluez/hci0/dev_FD_67_23_AA_07_AF
[...lots of devices...]
DEBUG:bleak.backends.bluezdbus.client:Connecting to BLE device @ FD:67:23:AA:07:AF with hci0
DEBUG:bleak.backends.bluezdbus.client:Connection successful.
DEBUG:bleak.backends.bluezdbus.client:Get Services...
DEBUG:bleak.backends.bluezdbus.client:
Primary Service
	/org/bluez/hci0/dev_FD_67_23_AA_07_AF/service004b
	0000fcba-0000-1000-8000-00805f9b34fb
	Vendor specific
[...lots of characteristics and services...]
Device connected?: True
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_FD_67_23_AA_07_AF/service0023/char0026, domain: org.bluez.GattCharacteristic1, body: {'Value': [68, 70, 85, 61, 49]}
DEBUG:bleak.backends.bluezdbus.client:Read Characteristic 00002a24-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_FD_67_23_AA_07_AF/service0023/char0026: bytearray(b'DFU=1')
bytearray(b'DFU=1')
disconnecting...
DEBUG:bleak.backends.bluezdbus.client:Disconnecting from BLE device...
DEBUG:bleak.backends.bluezdbus.client:Removing rule PropChanged, ID: 1
DEBUG:bleak.backends.bluezdbus.client:Disconnecting from BLE device...
```
In bluetoothctl I can see that the device is correctly disconnected:
```
[CHG] Device FD:67:23:AA:07:AF Connected: yes
[CHG] Device FD:67:23:AA:07:AF ServicesResolved: yes
[CHG] Device FD:67:23:AA:07:AF ServicesResolved: no
[CHG] Device FD:67:23:AA:07:AF Connected: no
[bluetooth]#
```

BillMicro, 2020-06-17:
As you can see from bleak/bleak/backends/client.py:
```python
async def __aexit__(self, exc_type, exc_val, exc_tb):
        await self.disconnect()
```
disconnect() is called when leaving the BleakClient with-block, there is therefore no need to call client.disconnect() inside this block.

hbldh, 2020-06-24:
This will be addressed in version 0.8.0. Thank you for reporting this!

paoloaveri, 2020-06-27:
Great, thank you!

dlech, 2020-06-27:
Probably better to not close this yet or it might be forgotten before it is fixed.

commit c267f9a4 (2020-06-30): Improved handling of txdbus connection object at disconnect in BlueZ backend.

Fixes #216
Fixes #219
Fixes #221
closed 2020-06-30 by hbldh